espressif/esptool #19: "Failed to connect"

InnovativeIdeas, 2015-01-20:
I used esptools about 2 weeks ago to flash  up to the latest Bin images ( from espressif github).  in that I went from V0018 0.9.2 to V 020 0.9.4  all worked fine no issues

I now find I cannot communicate with the ESP unit using ESPtools 

I m trying CH_PD high, ( direct to 3V3) , pulling GPIO0 low and reseting to enter boot loader mode

I wonder did the boat loader get changed  and this is why I can't connect

pihomeserver, 2015-01-24:
Same issue here. Can't get it work :-(
```
./esptool.py  -p /dev/cu.usbserial write_flash 0x000000 AI-v0.9.5.0\ AT\ Firmware.bin
Connecting...
Traceback (most recent call last):
  File "./esptool.py", line 408, in <module>
    esp.connect()
  File "./esptool.py", line 143, in connect
    raise Exception('Failed to connect')
Exception: Failed to connect
```

mizdler, 2015-01-27:
Make sure that you put your device in flash mode.

pihomeserver, 2015-01-27:
GPIO0 is Linked to the ground

InnovativeIdeas, 2015-01-27:
No, theres several issues with esptools/ESP. Ive gone through the software and a logic analyser

The first is that esptools  always expects a perfectly formed packet, i.e. it does not throw away characters  until it reaches the \0C, hence on some versions , the ESP does not auto baud perfectly and returns a junk character. This gives the an esptools error .  

IN other cases the ESP  actually boot up in  funny baud rate approx 74K.  if it doesn't auto baud properly,  you will get fail to connect messages,

The ESP boot loader isn't well implemented

InnovativeIdeas, 2015-01-27:
I think everyone on the planet knows about Gpio0 at this stage !

mizdler, 2015-01-27:
Yeah, but I had the same problem and it turns out that after connecting GPIO0 to ground you must restart the ESP. I mean just unplug it from the power and plug it again.

InnovativeIdeas, 2015-01-27:
Thats what I do all the time, but occasionally I find the ESP does not auto baud correctly and generates framing errors on the serial line, this confuses ESPtools.  

I have it working reasonably reliably , I modified esptools to cut down on the sync and also to scan for header rather then just expect it. I'll  tidy the code when I get a chance  and maybe post it for others, but Im actually writing a native Os/x flash utility at the moment .

pihomeserver, 2015-01-27:
Thanks that's usefull

pihomeserver, 2015-01-27:
That is *really* usefull. I will try this way and check if it works

themadinventor, 2015-01-27:
@InnovativeIdeas :
> I modified esptools to cut down on the sync and also to scan for header rather then just expect it.

Nice tweak. Scanning for `0xc0` only works during the connect phase; if we lose sync later during a session we'll probably discard the response.
You wouldn't mind sending a PR? Otherwise I'll implement something similar when I get time.

alesrebec, 2015-02-07:
Same issue for me.
Currently my ESP8266 is bricked.

```
sudo ./esptool.py -p /dev/ttyUSB0 write_flash 0x000000 ../lua/0x00000.bin 
Connecting...
Traceback (most recent call last):
  File "./esptool.py", line 471, in <module>
    esp.connect()
  File "./esptool.py", line 149, in connect
    raise Exception('Failed to connect')
Exception: Failed to connect
```

Tried with different bout settings to but all with same exception. 
Any clues how I can re-flash my ESP?

alesrebec, 2015-02-07:
Ok, found out my problem. Maybe it could solve yours too.
So I tried with a fresh new ESP8266 (they are so cheap that you gotta have at least 5 of them at home) and the new esp was getting the same error. 
Well, the problem was in USB UART's GND. I have a different power source for esp's 3.3V and was using only TX/RX for flashing. Forgot to plug UART's GNT on the same GND as esp's.

themadinventor, 2015-02-07:
@alesrebec: Good to hear that you solved it!

It isn't possible*, as far as I can tell, to completely brick the ESP8266. The bootloader resides in ROM and executes before the SPI flash is even initialized.

(The attentive hacker would point out that one could flash an image that specifies an invalid SPI flash clock rate and/or interface width, but looking at the routine at `0x40004644` in ROM, it looks like it'll still use a hard coded configuration `0x5` when booting from UART.)

(* Yeah, it *might* be possible to trash the OTP ROM containing the MAC address, but not using esptool.)

nfriedly, 2015-02-11:
@alesrebec Thanks! I've been trying to get this to work for like 3 hours before I found that - connecting my ftdi chip's ground to my external power supply's ground fixed it perfectly.

Jeshwanth, 2015-02-14:
Hi,
Same issue here... Can anybody fixed this issue ? I tried a lot of times, but no luck.

themadinventor, 2015-02-15:
@Jeshwanth Did you find a solution? I think I saw you on `#esp8266`, though a couple of hours too late...

Jeshwanth, 2015-02-15:
@themadinventor yes found solution.. Thats was the mistake in my circuit :). Forgot to update here... Thank you :)

jlz3008, 2015-02-21:
I solved the problem with a lower speed
```
jlz@isengard:~$ cp ../../firmware/AI-v0.9.5.0\ AT\ Firmware.bin f.bin
jlz@isengard:~$ ./esptool.py --baud 115200 --port /dev/ttyUSB0 write_flash 0x000000 f.bin 
Connecting...
Traceback (most recent call last):
  File "./esptool.py", line 377, in <module>
    esp.connect()
  File "./esptool.py", line 151, in connect
    raise Exception('Failed to connect')
Exception: Failed to connect
jlz@isengard:~$ ./esptool.py --baud 9600 --port /dev/ttyUSB0 write_flash 0x000000 f.bin 
Connecting...
Erasing flash...
Writing at 0x0007ef00... (100 %)

Leaving...
jlz@isengard:~$
```

gSTP, 2015-03-27:
thanks all, problem solved.

all hardware wired up nicely still produced the ugly "failed to connect" error. Just cranked up the external power for arduino nano breakout board to 10V, and that was it:

Connecting...
Erasing flash...
Writing at 0x00000400... (100 %)

Cheers all around  :-)

ColinMiles, 2015-07-30:
@alesrebec old discussion just just have to thank you for suggesting connecting the ground of the FTDI to ground of ESP!  THANK YOU!!  Nowhere had I seen that suggestion other than here.

nfriedly, 2015-07-31:
I'm on a new module (the custom one from electrodragon) and running into this again. It has the Ai Thinker AT command set preinstalled and that works fine; I can interact with it in esplorer and such. But, I can't flash anything new to it.

This build spits out a bit of debug info at 74880 baud when it first boots up and then it switches to 115200 for the "regular" command set. I'm suspecting that this switchover is what's making esptool choke - it fails to connect regardless of what baud I set it to:

```
Nathans-MBP:esptool nfriedly$ ./esptool.py  -p /dev/cu.SLAB_USBtoUART write_flash 0x00000  ~/Downloads/nodemcu-master-16-modules-2015-07-30-19-07-36-integer.bin 
Connecting...
Traceback (most recent call last):
  File "./esptool.py", line 505, in <module>
    esp.connect()
  File "./esptool.py", line 158, in connect
    raise Exception('Failed to connect')
Exception: Failed to connect

Nathans-MBP:esptool nfriedly$ ./esptool.py -b 74880 -p /dev/cu.SLAB_USBtoUART write_flash 0x00000  ~/Downloads/nodemcu-master-16-modules-2015-07-30-19-07-36-integer.bin 
Connecting...
Traceback (most recent call last):
  File "./esptool.py", line 505, in <module>
    esp.connect()
  File "./esptool.py", line 158, in connect
    raise Exception('Failed to connect')
Exception: Failed to connect

Nathans-MBP:esptool nfriedly$ ./esptool.py -b 115200 -p /dev/cu.SLAB_USBtoUART write_flash 0x00000  ~/Downloads/nodemcu-master-16-modules-2015-07-30-19-07-36-integer.bin 
Connecting...
Traceback (most recent call last):
  File "./esptool.py", line 505, in <module>
    esp.connect()
  File "./esptool.py", line 158, in connect
    raise Exception('Failed to connect')
Exception: Failed to connect

Nathans-MBP:esptool nfriedly$ ./esptool.py -b 9600 -p /dev/cu.SLAB_USBtoUART write_flash 0x00000  ~/Downloads/nodemcu-master-16-modules-2015-07-30-19-07-36-integer.bin 
Connecting...
Traceback (most recent call last):
  File "./esptool.py", line 505, in <module>
    esp.connect()
  File "./esptool.py", line 158, in connect
    raise Exception('Failed to connect')
Exception: Failed to connect
Nathans-MBP:esptool nfriedly$ ./esptool.py -b 1200 -p /dev/cu.SLAB_USBtoUART write_flash 0x00000  ~/Downloads/nodemcu-master-16-modules-2015-07-30-19-07-36-integer.bin 
Connecting...
Traceback (most recent call last):
  File "./esptool.py", line 505, in <module>
    esp.connect()
  File "./esptool.py", line 158, in connect
    raise Exception('Failed to connect')
Exception: Failed to connect
```

I tried both connection the DTR/RST pins, just pressing the RST/PROG buttons, doing both, etc. Nothing seems to work. Ground is connected correctly and I'm powering the module directly from my USB's 5v line (it has an onboard regulator).

But, like I said, I *think* the issue is the sudden baud rate switch - @InnovativeIdeas touched on it, but I don't think anyone's explicitly mentioned that it starts at one baud and then switches to another within a second or so.

If it helps, here's what the initial output at 74880 baud looks like:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 1396, room 16 
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4 
tail 8
chksum 0xc0
csum 0xc0

2nd boot version : 1.4(b1)
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000
```
and then it switches to 115200 and outputs something along these lines:

```
Ai-Thinker Technology Co. Ltd.

ready
WIFI CONNECTED
WIFI GOT IP
```

bislinux, 2015-08-11:
Just few bullet points to sum up
1. Connect EN to 3.3v
2. Connect GPIO0 to GND
3. Connect GPIO15 to GND to flash
4. Remember to connect the reference GND to the GND of esp8266 
5.Lastly correctly match the tx and rx lines

To flash the firmware
./esptool.py --baud 115200 --port /dev/ttyUSB0 write_flash 0x000000 ../nodemcu-flasher-master/Resources/Binaries/nodemcu_integer_0.9.5_20150318.bin

hope it helps

minouye, 2015-09-20:
@nfriedly Thanks for the workaround! Using NodeMCU Flasher inside a Windows VM worked for me as well--played nice with esptool after that.

ToM-Korn, 2015-10-12:
I had the same problem...
I had a LED at GPIO2 (with a 1k Resistor) just for testing.
One problem seems to be, that if GPIO2 is not high at the moment of programming it does not work. 
Having GPIO2 floating (not connected to anything) solved it. But as some docs say it would be better to have it high (3.3V) -> see here: 
http://www.esp8266.com/wiki/doku.php?id=esp8266_gpio_pin_allocations
Note :- GPIO2 It is considered safer to pull it high with a resistor on boot rather than leave it floating to avoid possible chip damage if pin is pulled high by your program but also see http://www.esp8266.com/viewtopic.php?f=6&t=3862&p=22524#p22524. 

So I'd recommend:
1) check if your pins are set like in the table 
2) check if the GND of your FTDI is connected to GND of ESP8266 (otherwise the would have different potentials)
3) set your baudrate to 115200 (worked fine for me all the time)

itcarenko, 2015-11-03:
Hi!

I had the same problem. But I solved it by also connecting CH_PD pin to the VCC

angelvil, 2015-12-13:
Hi. I have the same problem, but after to connect gnd to ftdi say: faled to connect. any opinion?
[image]

projectgus, 2016-02-25:
Closing this issue as the README now covers most of the likely problems.

Please open a new issue if you've tried the troubleshooting steps in the README and nothing works.

closed 2016-02-25 by projectgus

MuhsinFatih, 2016-03-03:
I had been trying to fix this problem with my esp8266 12e for days. I tried a lot of solutions to fix it. Tried esptool, arduino ide, different baud options etc all kinds of stuff. Just now figured out that it was very stupidly simple solution for me. Just plugged it into a windows machine (I had been using a macbook) and it worked with arduino ide with no change on the default esp programming options. In case it helps, i will tell a little about my setup: I have arduino with the processor removed so I use it for its serial pins. esp's rx goes to arduino's rx (with a voltage divider of course) and tx to tx (no divider). I have the reset line of esp connected to 3.3v and a button to short it to gnd (apologies for ignorance, is this called a pullup resistor?). I guess you know the rest of the connections if you're at this stage anyways.

projectgus, 2016-03-03:
Thanks for reporting your bug and providing so much detail.

You didn't make a mistake, there's a known bug with esptool.py when using an Arduino as the serial converter (see #82) and I think you've hit on the same issue.

There's no problem commenting on closed issues, however I think #82 is the correct issue in this case. If you don't mind, please subscribe to that issue (there's a button in the right-hand column) and you'll be notified once there's any update including a fix to test out (hopefully soon.)

MuhsinFatih, 2016-03-04:
Thanks! I subscribed. I will be waiting for the solution. Good luck!

julr08, 2016-04-13:
Thanks a lot, my problem was the Ground must be the same. the solution for me.

BlackHALk, 2016-04-22:
My blue light wont get off and random characters are showing in the serial monitor, need help ...

projectgus, 2016-04-24:
@BlackHALk , please go through the troubleshooting suggestions here: https://github.com/themadinventor/esptool#troubleshooting

ukrsms, 2016-05-27:
reading these issues I hoped I'd find the one which help me, but unfortunately any of them was the one.

MuhsinFatih, 2016-05-27:
@ukrsms off topic: man any doesn't mean none. This is mistaught by english teachers 99.999% of the time :D Just as a friendly reminder ;)

brynnb, 2016-07-17:
Having the same issue as @nfriedly - I have an ESP-12-E and communicating with minicom is working fine, but esptools won't connect. Unfortunately I'm still waiting on my USB/TTL converter so using the NodeMCU flasher isn't an option.

projectgus, 2016-07-18:
@brynnb : nfriedly's problem turned out to be the ground connection from the USB/serial converter. How are you connecting the ESP-12-E now if you don't have a USB/TTL converter on hand?

The README has sections on what connections need to be made, and some steps to follow if things aren't working correctly: https://github.com/themadinventor/esptool#entering-the-bootloader

brynnb, 2016-07-18:
@projectgus I have it wired directly to the GPIO pins of my Raspberry Pi. 

Fairly sure I had wiring correct, I will double check in the future using that guide since I didn't see it before. But at a glance it looks to be the same advice I found/followed elsewhere.

ReinVelt, 2016-12-11:
I had the same problems, tried everything and nothing worked.
After i **replaced the USB cable** everythink worked perfectly!

SteveBaker, 2017-02-05:
I've been working with dozens of ESP 12F boards without problems for six months or so now - but I just got a new batch of boards - and now I get  the dreaded "Failed to connect" problem.   I can talk to the board with "screen /dev/ttyUSB0 115200" - and the "AT" commands seem to be working.  Here is the AT+GMR version command's output for this new batch of boards:

  AT version:1.2.0.0(Jul  1 2016 20:04:45)
  SDK version:1.5.4.1(39cb9a32)
  Ai-Thinker Technology Co. Ltd.
  Dec  2 2016 14:21:16

That last line suggests that we have a new build here.  (Dec 2016!)

The latest esptool.py (just downloaded) says:
===============================================================
esptool.py v2.0-beta1
Creating image for ESP8266...
../tools/esptool.py --port /dev/ttyUSB0 --baud 460800 write_flash   0x00000 About-0x00000.bin 0x40000 About-0x40000.bin
esptool.py v2.0-beta1
Connecting...........................................

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
Makefile:12: recipe for target 'flash' failed
make: *** [flash] Error 2
===============================================================
The exact same command works just perfectly with older copies of the same ESP-12F boards.

SteveBaker, 2017-02-05:
I've been spending the last 24 hours fighting a "Failed to connect" problem - I had a new batch of ESP 12F's arrive - and when I tried to program one, I got the error.  According to the AT-GBV command, it has a newer version of the boot loader than any of my older ESP chips (Dated Dec 2nd 2016!) - and of course I naturally assumed that they broke something!  But after trying my best to diagnose it by hacking around in esptools.py (and my Python skills seem to have rusted rather badly - so that didn't go well) - I eventually gave up and decided to try a second board from the same batch...lo and behold, it worked perfectly...so it looks like I had a bad ESP board.  Fortunately, at $2 - who gives a damn!

Good news!   The Dec 2nd build works OK!

Moral of the story: Before assuming the worst - try another ESP board!  :-)

iampt, 2019-03-21:
Flash any firmware to ESP8266 using following steps..
1. load blank sketch to arduino (without any wiring)
2. unplug the usb and do this connection
ESP --> ARDUINO
GPIO0 -> GND
GND -> GND
RX -> RX
TX -> TX
3V3/VCC -> 3.3V
CH_PD/EN -> 3.3V

plug in the usb to the power source.

download the official flasher from this link, unzip and open the app https://drive.google.com/file/d/0B3dUKfqzZnlwVGc1YnFyUjgxelE/view

download the firmware from this link, unzip it and open it in flasher app
www.electrodragon.com/w/File:At_firmware_bin1.54.zip

make sure to check and select accurate com port in flasher
click on download

That's it, you may open the serial monitor now and perform AT commands.

HOPE THIS HELPS. THANK YOU! :)

aog1971, 2019-04-18:
> Flash any firmware to ESP8266 using following steps..
> 
> 1. load blank sketch to arduino (without any wiring)
> 2. unplug the usb and do this connection
>    ESP --> ARDUINO
>    GPIO0 -> GND
>    GND -> GND
>    RX -> RX
>    TX -> TX
>    3V3/VCC -> 3.3V
>    CH_PD/EN -> 3.3V
> 
> plug in the usb to the power source.
> 
> download the official flasher from this link, unzip and open the app https://drive.google.com/file/d/0B3dUKfqzZnlwVGc1YnFyUjgxelE/view
> 
> download the firmware from this link, unzip it and open it in flasher app
> [www.electrodragon.com/w/File:At_firmware_bin1.54.zip](http://www.electrodragon.com/w/File:At_firmware_bin1.54.zip)
> 
> make sure to check and select accurate com port in flasher
> click on download
> 
> That's it, you may open the serial monitor now and perform AT commands.
> 
> if AT commands are not working then unplug usb from the power source, remove GPIO0 pin from GND and plug in the USB to power source. open serial monitor and recheck the AT commands.
> 
> HOPE THIS HELPS. THANK YOU! :)

Thanks this has worked for me, doing these steps I could put flash mode and put the firmwaere.

aog1971, 2019-04-21:
> Just few bullet points to sum up
>
> 1. Connect EN to 3.3v
> 2. Connect GPIO0 to GND
> 3. Connect GPIO15 to GND to flash
> 4. Remember to connect the reference GND to the GND of esp8266
>    5.Lastly correctly [...]

Thanks this has worked for me, doing these steps I could put flash mode and put the firmware

SirSeko, 2019-05-04:
> Just few bullet points to sum up
>
> 1. Connect EN to 3.3v
> 2. Connect GPIO0 to GND
> 3. Connect GPIO15 to GND to flash
> 4. Remember to connect the reference GND to the GND of esp8266
>    5.Lastly correctly [...]
Thank you!! I´ve been troubleshooting my unit for a month and finally found this solution!

flaviostutz, 2020-07-05:
> Thats what I do all the time, but occasionally I find the ESP does not auto baud correctly and generates framing errors on the serial line, this confuses ESPtools.
>
> I have it working reasonably [...]

Have you open sourced it?!

dimitre, 2020-08-11:
I can always write only once per chip. to write again in the same chip I have to unplug USB or the chip and plug it again. If I don't do that I get the infamous
```
Connecting........_____....._____....._____....._
```

Odissine, 2020-10-04:
I still have the same issue to ... and to avoid buy a usb cable, is there a way to know if a Mini USB cable is for power only or for data too ... I think maybe the problem is here ?