espressif/esptool #21: PEP 394 - The "python" Command on Unix-Like Systems

gdamjan, 2015-02-07:
https://www.python.org/dev/peps/pep-0394/

It suggests to use the supported python version in the shebang, in this case it should be
`#!/usr/bin/env python2`

themadinventor, 2015-02-08:
The shebang line stuff is a clusterfsck. `/usr/bin/env python2` does not work out of the box on f.ex. Mac OS 10.9 (at least on my system). Specifying `python2` is of course correct, but it'll always break on some systems, and still, citing PEP394:
> for the time being, all distributions should ensure that python refers to the same target as python2

I guess the best solution is to make esptool compatible with both Python 2 and 3, keeping it `python`, and if not possible, switch to `python3` and force users to install it.

dpgeorge, 2015-02-08:
> I guess the best solution is to make esptool compatible with both Python 2 and 3

+1

gdamjan, 2015-02-08:
The other approach is to make esptool a proper package/module and register an `console_scripts` entry-point [1]. This way, when setup.py install it, it will create a proper script with the correct shebang for the installed python. 

[1] http://www.scotttorborg.com/python-packaging/command-line-scripts.html#the-console-scripts-entry-point

jiegec, 2015-02-23:
Or consider using #!/use/bin/env python2.7 ? Tested on Mac and Arch Linux

jiegec, 2015-02-23:
I also agree with @gdamjan

projectgus, 2015-08-31:
I've pushed an experimental branch that runs on Python 2 and 3.
https://github.com/themadinventor/esptool/compare/python2_and_3

I'm not 100% convinced this isn't making a maintenance trap for down the road, but the differences weren't too terrible. I haven't tested all operations, some things probably still don't work.

Any suggestions or testing is welcome, though.

dbrgn, 2015-10-18:
Python 2/3 support would be great :)

projectgus, 2016-04-21:
I recently became an Arch user myself, so I wrote up some tips/workarounds here:
https://github.com/themadinventor/esptool/wiki/Arch-Linux-Tips

(I'm using pip 'development mode' myself now.)

jrobeson, 2016-09-21:
@projectgus : I took a look at the branch.  I think it should start a little smaller.  Can we adapt the coding standards (defacto or otherwise) so that all byte literals are marked up with `b'`?  That'd go a long way to making 2/3 work and stay working.

projectgus, 2016-09-21:
@jrobeson that's a good idea! The python2_and_3 branch is probably so out of date it'll need to be started from scratch again, anyhow.

jeremyherbert, 2016-09-22:
I've just opened a pull request of this tool to which works on python2 and python3. I'd like to use micropython, and my feeble mind cannot handle running a script in python2 to install a python3 interpreter ;) It's here: https://github.com/themadinventor/esptool/pull/127

This should fix this issue.

Questions/comments/suggestions welcome.

projectgus, 2016-12-19:
Current master branch now supports Python 2.7 & Python 3.4+. This will be in the forthcoming v1.3 release. Please report any bugs as new issues.

Thanks to @jeremyherbert for getting the ball rolling. Hurrah for closing issues before they reach their second birthday.

closed 2016-12-19 by projectgus