espressif/esptool #22: 'Failed to connect' with multiple messages

jqdoumen, 2015-02-15:
Downloaded esptool, connected GPIO2 to 3.3V and GPIO0 to ground, booted up and executed the esptool as follows (with resulting output). 

$ sudo esptool.py --port /dev/tty.usbmodem1451 write_flash 0x000000 Firmware.bin 
Connecting...
Traceback (most recent call last):
File "/usr/local/bin/esptool.py", line 5, in <module>
pkg_resources.run_script('esptool==0.1.0', 'esptool.py')
File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 492, in run_script
self.require(requires)[0].run_script(script_name, ns)
File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 1357, in run_script
exec_(script_code, namespace, namespace)
File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 47, in exec_
exec("""exec code in globs, locs""")
File "<string>", line 1, in <module>
File "/Library/Python/2.7/site-packages/esptool-0.1.0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 471, in <module>

File "/Library/Python/2.7/site-packages/esptool-0.1.0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 149, in connect

Exception: Failed to connect

Given the error messages, and not a trace of similar issues reported by others, is this due to the code? The 'failed to connect' could be interpreted by a wrong connection on the board or so, but the error messages makes me assume there is something not right with the esptool.

themadinventor, 2015-02-15:
Seems to me that the stack trace is just longer because esptool is invoked through a stub generated by setuptools; the last two rows or so in the trace are the same. It is perhaps time to implement proper error messages... ;)

jqdoumen, 2015-02-15:
Does this mean you suspect that it is not the code but something with the ESP8266 or the board is wrong?

Some extended error messages might help indeed ... :-)

themadinventor, 2015-02-15:
Most if not all issues reported around this error have turned out to be about bad wiring and/or power supply.
However, letting esptool control the reset and boot mode pins through RTS/CTS *might not* work on some platforms, so controlling those pins "in hardware" is probably a good troubleshooting step, but it sounds to me that you're already doing that.

jqdoumen, 2015-02-16:
Been revising the whole setup ... I was using the 3.3 V output of the Arduino Uno, which seemed to work even though a lot of people claim it doesn't. But just to make sure, I got it now on 5V converted tot 3.3 through an L78L33 converter. The result is:

VCC: 3.3V
Reset: left open
CH_PD: 3.3V
TX: directly to RX pin on Arduino
GND: to -
GPIO2: 3.3V
GPIO0: to -
RX: to TX of Arduino through another L78L33

With the GPIO2 and GPIO0 left open, I got the blue flashing. With both connected as explained above, not blue flashing, so I guess correctly configured. 
Still, the errors I get with the esptool remain the same ... 

:-(

themadinventor, 2015-02-16:
I wouldn't recommend using a voltage regulator for level translation, it is probably to slow and has a high forward voltage. If you don't have a 3.3v FTDI cable/module or level translation chip at hand, try making a voltage divider from two resistors.
Try loopbacking your serial interface by shorting Tx ro Rx, connecting with a terminal emulator (at 115kbaud) and see if you can receive the characters you type ungarbled.

I don't actually own an arduino, otherwise I would have tried to reproduce your setup :/

Hope this helps :)

a-pedde, 2015-02-17:
qdoumen each L78L33 (like most other Low Dropout regulator) needs 2 capacitors (e.g. http://www.st.com/web/en/resource/technical/document/datasheet/CD00000446.pdf Page 3 test circuits).
Output current up to 100 mA is not enough to power the esp8266 (maybe your L78L33 is from an other vendor then ST, please check the datasheet if you will get 200+ mA output).
don't use the L78L33 for the RX: to TX line, better use a voltage divider from two resistors.

arduino-resistor-example:
http://www.instructables.com/id/Cheap-2-Way-Bluetooth-Connection-Between-Arduino-a/?ALLSTEPS

Don't use the 5V like shown in the example ;) 

1. Which arduino board do you use ( i guess arduino uno with(out) atmega328p?)

2. Do you use the arduino board just like a FTDI-replacement?

3. Is the atmega328p disconnected from the board? (This should be done if you only use the arduino-board as an FTDI-replacement)

jqdoumen, 2015-02-17:
Hi themadinventor and a-pedde, thanks for you giving a helping hand, even though this is most likely not the proper place for it ...

I'll possibly pose some more novice questions ... 

I've tried to create the voltage dividers, both for the RX-TX line, and the power. Before, when using the pure 3.3V of the Arduino, or the L78L33, the ESP8266 was functional, for several days (OK, it restarted a couple of times, but most of the time was working properly). After I screwed up with the on-line firmware flash, the chip still had the red light on (in firmware upgrade modus) or the red light on and the blue light flashing (in normal modus). With the voltage divider setup now, there is nothing, not lights whatsoever. 

Setup is : grnd - 2KO - split off - 1KO - 5V (from the Arduino). 

What I couldn't find out is wether the split off can contain only one connection or more. That is, should I create separate voltage dividers for VCC, GPIO2 and CH_PD?

Yes, I'm using the Uno, with the atmega328p chip. I actually don't feel comfortable removing that chip, so if the flashing should be done without, then I'd rather go for a seperate FTDI cable.

Regarding the L78L33, no idea what the brand name is ..

/Users/jandoumen/Downloads/IMG_0168-1.jpg

a-pedde, 2015-02-19:
you shouldn't use the resistors to get 3,3v for VCC!
its only for the tx/rx lines.
you can't use the arduino-board to programm the ESP without disconnecting the atmega-chip.

try to get a proper LDO (LM 1117 T3,3, TS 2940 CZ33, maybe your L78L33 is ok if it has more then 200mA output current) + 2 caps shown in the datasheets of the LDO you get.

i dont think the 3,3v from arduino would be enough to connect to wifi (maybe this was the reason for the brick while you tried to flash your module via WiFi?)

The 3,3v out of the cheap FTDI/SILABS/... USB2UART adapters aren't enough for proper usage with the esp8266, too

zendes, 2015-02-19:
I'm a big fan of eBay for these kind of parts if you don't want to make them yourself. Something like www.ebay.com/itm/291156034828 coupled with a 1A usb phone/tablet charger will get you a stable power souce.

Baael, 2015-11-12:
Had similar problems with UART. 
I needed to connect reset PIN to reset on UART - it solved that problem. 
For one of three same modules I had to add additional power source, two of them work nice with Arduino/UART only (ESP8266 esp-01)

projectgus, 2016-02-25:
Closing this issue as the README now covers most of the likely problems.

Please open a new issue if you've tried the troubleshooting steps in the README and nothing works.

closed 2016-02-25 by projectgus