espressif/esptool #23: DTR stops too soon

myplacedk, 2015-02-22:
When I flash my ESP8266, I must keep GPIO0 low during the entire procedure. If I disconnect it during flashing, it immediately stops.

I'm now trying to let esptool handle RESET and GPIO0 with RTS and DTR. But esptool only enables DTR (which holds GPIO0 low) for 0.1 seconds.

By removing "self._port.setDTR(False)", esptool seems to work perfectly.

projectgus, 2015-08-30:
Can you give any more details about the setup (module/wiring) that requires GPIO0 to stay low? The ESP only checks GPIO0 when the initial reset occurs, and some wiring combinations may have problems if GPIO0 gets held low.

projectgus, 2016-02-25:
I'm going to close this as it's been awhle, but if you have any more details about what you think was unique with your wiring setup, please reopen.

closed 2016-02-25 by projectgus