espressif/esptool #25: Exception: Invalid response

Hukuma23, 2015-04-06:
Hello,

I tried to write_flash binaries to ESP-01 but every time got exception. Full log:
sudo ./esptool.py write_flash 0x00000 WiFiWebServer.cpp_00000.bin 0x40000 WiFiWebServer.cpp_40000.bin 
Connecting...
Erasing flash...
Traceback (most recent call last):
  File "./esptool.py", line 525, in <module>
    esp.flash_begin(blocks*esp.ESP_FLASH_BLOCK, address)
  File "./esptool.py", line 188, in flash_begin
    struct.pack('<IIII', size, num_blocks, ESPROM.ESP_FLASH_BLOCK, offset))[1] != "\0\0":
  File "./esptool.py", line 110, in command
    raise Exception('Invalid response')
Exception: Invalid response

MAC OS 10.10.2

The same binaries I can to write to flash from Windows computer, what's wrong with MAC? Also I can't upload compiled sketch from esp8266 arduino IDE on my MAC but there is no problem on Windows PC. Please help to understand.

icnagy, 2015-04-17:
Got the same problem on 10.10.3.

projectgus, 2016-03-04:
Sorry we never got back to you. I'm going to close this issue because it's been so long between comments.

There's now a list of troubleshooting suggestions in the README, please run through these if you're still having this problem:
https://github.com/themadinventor/esptool#troubleshooting

If it's still not working, please reopen the issue with some details about the esp8266 hardware configuration you're using, and output from a current version of esptool.py.

closed 2016-03-04 by projectgus