espressif/esptool #26: Support for NodeMCU-devkit 0.9 on Mac?

abl, 2015-04-12:
Hi! I've got a couple of NodeMCU Devkits of the 0.9 rev (so the CH340G bridge) and I'm using the signed driver from mac-usb-serial.com. Out of the box, esptool isn't working.

These changes seem useful:

    self._port.setRTS(True)
    self._port.setDTR(False)
    time.sleep(0.1)
    self._port.setRTS(False)
    self._port.setDTR(True)
    time.sleep(0.1)

Doing this causes the ESP to boot in to mode (1,6) - user firmware boot identifies itself as (3,6).

The boot header is ```ets Jan  8 2013,rst cause:2, boot mode:(3,6)``` so it's a somewhat older firmware. I can't find any docs on what the second number means, but all of my working devices show a 7...

Any thoughts? :)

abl, 2015-04-12:
Relying on USB for power, and connecting a different serial device, I boot in to (1,7) and was able to reflash without any issues. Even more mystified now...

MacDada, 2015-04-17:
* I've had AT commands working, but it stopped when I did `AT+CIUPDATE`.
* I can't get any CH340G driver to work on Yosemite.
* I installed ubuntu with vagrant+virtualbox to make sure no OSX–specific software problems.
* I power NodeMcu from Mac using built in USB.
* I "talk" with NodeMcu through Arduino Mega serial with it's Atmega chip shut down – it speaks "garbage" (because of `AT+CIUPDATE`).
* Flashing with `esptool` doesn't work.
* I tried both with and without voltage divider.

@abl I show my setup plus what works and what doesn't here (sorry for my bad english :P): https://www.youtube.com/watch?v=AibvuZ4YB7Y

abl, 2015-04-24:
@MacDada I connected a 3.3V FTDI USB->UART to the RX/TX lines and held down the flash button as I connected the onboard USB for power.

projectgus, 2015-08-30:
I'm going to close this as I think it should be fixed as of #30. Please reopen if the nodemcu 0.9 still doesn't seem to work.

closed 2015-08-30 by projectgus