espressif/esptool #27: Read boot header for better error messages

abl, 2015-04-12:
So the ESP8266 outputs a boot header at 77400 baud containing something like:

```ets Jan  8 2013,rst cause:2, boot mode:(3,6)```

According to https://github.com/esp8266/esp8266-wiki/wiki/Boot-Process#esp-boot-modes the first value in the boot mode is interesting.

GPIO15 | GPIO0 | GPIO2 | Value | Status
-- | -- | -- | -- | --
0 | 0 | 1 | 1 | UART
0 | 1 | 1 | 3 | SPI Flash
1 | X | X | >= 4 | SDIO

If, after reset, we encounter a boot mode that isn't 1, it's possible to raise an error there and tell the user the board was not put in to flash mode.

projectgus, 2015-08-30:
This seems like a really interesting idea, but I'm not sure exactly how it would work.

Do you mean we scan the output for the ROM boot mode string, and then if we see it then we can parse out and give an intelligent "your GPIO0 line is high" type error message?

The only problem I see is that esptool.py doesn't usually operate at the oddball 74800 baud rate, the default is 115200 and most people run even faster than this. In fact only about 30% of the adapters I have at hand can even use that baud rate (FT2232R can but cp2102, ch341 and avr usbserial all can't.)

So I think it would be tricky to parse this out while also trying to operate at a higher rate. Or am I misunderstanding how it would work?

crushedice2000, 2015-11-23:
I think that he suggest to open the connection at 74800 and once it gets the bootloader response, change the speed to 115200 (or whatever speed you want)

radimkarnis, 2020-12-08:
I’m closing this issue because it has been inactive for a few months and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-08 by radimkarnis