espressif/esptool #28: Yet Another "Failed To Connect"

mzcs, 2015-04-13:
Hi There :)

I've Just Got My Very First ESP8266 Set-up, as In The Circuit Schematic (Which is Set According To The "ESP Boot Modes" Table Here : https://github.com/esp8266/esp8266-wiki/wiki/Boot-Process ) :
[image]

I'm Able To Communicate With It - via GNU' Screen & Found That It's Current S/W Ver. is "ai-thinker.com v0.9.2.4".

Nevertheless, themadinventor' esptool - While Seem To Be Able To Communiction With The ESP8266 (It's Comms. LED Blinks & GNU' Screen Outputs Some Stuff), Can't Setup a Operable Connection To It & Only Outputs This :

hercules ~ # /opt/esptool.py -p /dev/ttyUSB0 -b 9600 flash_id
Connecting...
Traceback (most recent call last):
  File "/opt/esptool.py", line 471, in <module>
    esp.connect()
  File "/opt/esptool.py", line 149, in connect
    raise Exception('Failed to connect')
Exception: Failed to connect

---

Any Ideas... ? :)

MacDada, 2015-04-13:
Check this out: https://github.com/themadinventor/esptool/issues/22

I'm also having the "failed to connect" error, but might not be for the same reason (gonna explain after I make a diagram with my configuration).

projectgus, 2015-09-02:
@mzcs, did you figure this one out? The schematic looks right, often the problem is found by fiddling with wiring though.

It also may be worth trying (oddly) a higher baud rate, the default 115200 should be fine unless your wiring is particularly long, noisy or complex.

projectgus, 2016-02-25:
Closing this issue as the README now covers most of the likely problems.

Please open a new issue if you've tried the troubleshooting steps in the README and nothing works.

closed 2016-02-25 by projectgus