espressif/esptool #31: How to run a terminal without driving RTS ?

d-a-v, 2015-04-24:
Hi,

So far I was using a switch and a button to program the chip with esptool.
I decided to give a go with RTS(CH_PD) and DTR(GPIO0) so I can program the chip remotely, for dev in the sofa.

It works, but I can't figure out how to run a terminal, because it drives RTS so the chip is reset everytime.
I'm trying with minicom. On linux, without hard/soft flow control, it says "no such device or address", and with them the terminal starts but the chip is unresponsive (because RTS driven). On OSX, no matter what hard/soft flow control, the terminal starts and the chip is unresponsive. Of course it works (linux, osx) if I manually drive CH_PD to 3.3v.

I still can access to it through telnet, but this is not ideal for dev, I love the serial port :)

I'm really wondering how do you guys do.

Thanks

themadinventor, 2015-04-26:
Hi d-a-v,

I myself wrote a simple/stupid terminal app in Python that sets the status lines. A side effect is that it allows arbitraty baudrates (e.g 74400 used by the bootloader), which picocom (my otherwise preferred serial app) doesn't.
I must admit that the pin assignment is also arbitrary (actually inherited from the serial adapter I happened to have at hand) so I guess this scheme should be deprecated. The NodeMCU team have another solution for which support in esptool will be implemented/merged.

GeorgeHahn, 2015-04-26:
Do you know where the NodeMCU solution is documented?

themadinventor, 2015-04-26:
The schematic is in their repo https://github.com/nodemcu/nodemcu-devkit/blob/master/Documents/NODEMCU_ESP12.PDF
It shows both the discrete logic and the corresponding truth table.

closed 2015-08-30 by projectgus