espressif/esptool #33: Add workaround for a bug in SPIEraseArea

igrr, 2015-04-29:
```SPIEraseArea``` function in the esp8266 ROM has a bug which causes extra area to be erased.
If the address range to be erased crosses the block boundary (block is 16 sectors), then **extra** ```head_sector_count``` sectors are erased.
If the address range doesn't cross the block boundary, then **extra** ```total_sector_count``` sectors are erased.
Where ```head_sector_count``` is the number of sectors from the start of erased area to the block boundary, and ```total_sector_count``` is the total number of sectors to be erased.

Espressif added the workaround to their version of flash tools. The trick is to modify the erased size:
```
    """ Start downloading to Flash (performs an erase) """
    def flash_begin(self, _size, offset):
        old_tmo = self._port.timeout
        self._port.timeout = 10

        area_len = int(_size)
        sector_no = offset/4096;
        sector_num_per_block = 16;
        #total_sector_num = (0== (area_len%4096))? area_len/4096 :  1+(area_len/4096);
        if 0== (area_len%4096):
            total_sector_num = area_len/4096
        else:
            total_sector_num = 1+(area_len/4096)
        #check if erase area reach over block boundary
        head_sector_num = sector_num_per_block - (sector_no%sector_num_per_block);
        #head_sector_num = (head_sector_num>=total_sector_num)? total_sector_num : head_sector_num;
        if head_sector_num>=total_sector_num :
            head_sector_num = total_sector_num
        else:
            head_sector_num = head_sector_num

        if (total_sector_num - 2 * head_sector_num)> 0:
            size = (total_sector_num-head_sector_num)*4096
            print "head: ",head_sector_num,";total:",total_sector_num
            print "erase size : ",size
        else:
            size = int( math.ceil( total_sector_num/2.0) * 4096 )
            print "head:",head_sector_num,";total:",total_sector_num
            print "erase size :",size

        if self.command(ESPROM.ESP_FLASH_BEGIN,
                struct.pack('<IIII', size, 0x200, 0x400, offset))[1] != "\0\0":
            raise Exception('Failed to enter Flash download mode')
        self._port.timeout = old_tmo
```

Perhaps after some cleanup this fix should be applied here as well.
This issue causes the first 0x1000 to be erased in some cases when writing near the end of flash. In turn, this leads to a corrupted flash image when writing two binaries (0x00000 and 0x40000) separately.

scargill, 2015-05-09:
That is very informative. I've been having trouble with this since version 1.0 of the SDK and reversing the flashing order sorts it - but it would be nice if it worked properly...  unfortunately the code above... I have no idea where that goes, using Windows Eclipse and Makefile. Any idea?

projectgus, 2015-08-30:
Thanks @igrr for the heads-up on this. Have merged a fix just now as #53, is equivalent (I'm fairly sure) but cleaner code contributed by @jkent.

closed 2015-08-30 by projectgus