espressif/esptool #35: Invalid head of packet with ESP-01

Lenbok, 2015-05-03:
I could not flash firmware to my ESP-01 and was consistently getting failures like this:
````
$ sudo python esptool.py --port /dev/ttyUSB0  write_flash 0x00000 nodemcu_latest.bin
Connecting...
Erasing flash...
Writing at 0x00008400... (8 %) 
Traceback (most recent call last):
  File "esptool.py", line 536, in <module>
    esp.flash_block(block, seq)
  File "esptool.py", line 195, in flash_block
    struct.pack('<IIII', len(data), seq, 0, 0)+data, ESPROM.checksum(data))[1] != "\0\0":
  File "esptool.py", line 106, in command
    raise Exception('Invalid head of packet')
Exception: Invalid head of packet
````
After some googling, I found this http://www.esp8266.com/viewtopic.php?f=6&t=2791#p16082
which resolved the problem. You may want to consider incorporating it or something equivalent in the main version.

pietrushnic, 2015-08-17:
+1

Thanks for this hint.

projectgus, 2016-03-04:
This has been reported in a couple of places (like #67), but it seems like a side effect of power problems - the ESP8266 is browning out during flashing, by sending less data in a single packet then the brownout effect becomes less severe.

If you're sure that your ESP-01 is powered properly (for instance, it runs 100% stable while connected to WiFi) and you still need this modification in order to flash it, please reopen the issue.

closed 2016-03-04 by projectgus

ceremcem, 2016-04-03:
I got the same error just because I left `/dev/ttyUSB0` port open in another process accidentally. Returning users may want to check this parameter too.

Trospector, 2017-12-17:
+1 ceremcem, I had the same problem and closing Arduino IDE which had the port open solved the problem.

TAHiRLI, 2021-11-24:
@Lenbok  if you are using arduino to flash the esp, try to upload blank code to your arduino and then flash it