espressif/esptool #36: "Failed to connect" but nodemcu-flasher worked

stheophil, 2015-05-09:
I tried to use the esptool for quite a while last night. I have an [ESP8266 201](http://smarpl.com/content/esp8266-esp-201-module-first-impressions) with the nodemcu firmware. I wanted to flash the latest nodemcu firmware on my Mac and only got the "Failed to connect" error message.

I tried setting the baud rate to some specific value, no change. 
I tried it with GPIO2 pulled up and floating, no change. 
I first powered the board from the USB-serial adaptor and then switched to a breadboard power supply (that ultimately is also powered by the USB port course), no change.

I finally installed a Windows trial version in Virtual Box and tried the nodemcu flasher, worked like a charm. So I suppose my wiring is OK. I have a cheap CH341 chipset USB-serial adapter, if that makes a difference. 

Is there any way I could provide some debug information that would make it easier to solve this issue?

jeremywho, 2015-05-11:
I'm having the same problem.  On OS X, I get the failed to connect error.  On Windows I can use nodemcu flasher no problem.

stheophil, 2015-05-11:
I’ve traced some information from the esptool.py tool today. It seems upon sending the sync message, I get no response at all. I can see the bootloader message at 75000 baud, so again, wiring etc are fine. 

I’ve read [elsewhere of problems with CH340 serial chips, Mac OS X and esptool](http://blog.squix.ch/2015/02/esp8266-nodemcu-dev-kit-review.html <http://blog.squix.ch/2015/02/esp8266-nodemcu-dev-kit-review.html>) that simply went away when using another serial adaptor. I’ve ordered an FT232-based adapter and will report back with the results.

stheophil, 2015-05-15:
The problem seems to be the CH341 chipset or the Repleo driver. I’ve bought the adafruit FTDI friend and it works without any issues.

LucasG2000, 2015-05-16:
Just a note, commenting the Exception in the connect method of esptool.py is enough for me for burning the firmware with a CH341 chipset under OS X. Leave like this https://github.com/themadinventor/esptool/blob/master/esptool.py (line 157):

```
    def connect(self):
        print 'Connecting...'

        for _ in xrange(4):
            # issue reset-to-bootloader:
            # RTS = either CH_PD or nRESET (both active low = chip in reset)
            # DTR = GPIO0 (active low = boot to flasher)
            self._port.setDTR(False)
            self._port.setRTS(True)
            time.sleep(0.05)
            self._port.setDTR(True)
            self._port.setRTS(False)
            time.sleep(0.05)
            self._port.setDTR(False)

            self._port.timeout = 0.3 # worst-case latency timer should be 255ms (probably <20ms)
            for _ in xrange(4):
                try:
                    self._port.flushInput()
                    self._port.flushOutput()
                    self.sync()
                    self._port.timeout = 5
                    return
                except:
                    time.sleep(0.05)
            # this is a workaround for the CH340 serial driver on current versions of Linux,
            # which seems to sometimes set the serial port up with wrong parameters
            self._port.close()
            self._port.open()

        #Comment this line (157)
        #raise Exception('Failed to connect')
```

projectgus, 2015-09-02:
There are reports in the nodemcu thread that getting the latest CH341 OS X driver fixes this problem.

Is anyone still seeing this problem with the latest CH341 drivers and the latest esptool.py?

InnovativeIdeas, 2015-09-17:
I revisited the  python code  because using OS X and a new ESP-12 module, esptool.py repeatedly  couldn't connect.  Im using a cheapo Profilic usb to TTL

The esptool suffers from a MAJOR flaw in the sync code, because it makes no allowances for ;

(a) The first SYNC command, will result in junk  being sent to the serial port as the ESP autobauds, Hence  esptool will never see a valid Header byte, and will constantly fail . The second SYNC command will be at the correct baud rate  ( Theres no need  for the  multiple SYNC after that )

(b) certainly on my ESP-12 out of the box, the response to the SYNC command is a series of  SLIP packet not one, i.e. about a dozen response commands. This confuses  esptools 

These I fixed  by various hacks , ( Im not proud )

I also found that with a 2 wire TTL , no handshake, that a small time.sleep(0.01) delay had to be placed before any command write to the serial port.  It looks like pyserial gets corrupted if you make back to back command writes and the second and subsequent commands never get sent.

my version , running under OS X now runs upto  115200, and connects first time , every time.

just be aware, that the existing code simply cannot function reliably as written certainly under OS X ( 10.9 etc ) 

other then that , its fine

sguilly, 2015-09-20:
Hi InnovativeIdeas,

Could you share your fix code ?

Thanks,

projectgus, 2016-02-25:
I'm going to close this issue as I think #65 fixed this sync issue among others.

If you're still seeing sync errors with current esptool.py, please open another issue with the details.

closed 2016-02-25 by projectgus