espressif/esptool #40: Please support the new NodeMCU V1.0 with DIO mode

squix78, 2015-05-15:
See https://github.com/nodemcu/nodemcu-firmware/issues/412 for the problem.
Flashing the module works with dio mode and 32M but in the LUA firmware accessing the Flash storage doesn't work. If using the same module with the latest espressif tool from 
http://bbs.espressif.com/viewtopic.php?f=5&t=433 works just fine.

I described the workaround here:
http://blog.squix.ch/2015/05/esp8266-flashing-nodemcu-v10-on-mac-os.html

Please support this mode in the esptool.py

vowstar, 2015-05-16:
DIO write data 01 FF FF FF AA 55 AA 55 01 00 00 00 FF FF FF FF 1C 00 00 00 FF FF FF FF F2 00 00 00 FF FF FF FF to header of last 1024Bytes of flash(If using 1MBytes flash it is 0xFF000). 
Please change this data when using DIO options, Thanks.

creationix, 2015-05-26:
Is someone working on this?  I don't quite understand @vowstar's comments, but I would love a working way to build nodemcu in linux and then flash to my nodemcu devkit on the same device.  It's a pain to transfer the build files to windows every time I want to flash.

squix78, 2015-05-26:
I have two notes to your point @creationix: 
1. a Linux user could most probably use the espressif tool together with wine. It worked for me under Mac OS X and I don't see why it shouldn't under Linux (see http://blog.squix.ch/2015/05/esp8266-flashing-nodemcu-v10-on-mac-os.html)
2. The Arduino IDE for ESP8266 seems to be fixed. Their bult-in esptool for upload works now for the NodeMCU V1.0. I tested it yesterday...

vcarriere, 2015-06-16:
I have found that the problems seems to be in the flash_begin function.

The problem seems to be at the line 197 in esptool.py

Espressif : 
struct.pack('<IIII', size, 0x200, 0x400, offset))[1] != "\0\0":

Esptool : 
struct.pack('<IIII', size, num_blocks, ESPROM.ESP_FLASH_BLOCK, offset))[1] != "\0\0":

In the esptool.py from themadinventor the 0x200 is replaced with num_blocks and it causes it not to work.

Am I missing something here? I’ve been trying to make this work for a week (first time working with ESP).

Please just let me know if this is the problem and explain.

closed 2015-07-08 by themadinventor