espressif/esptool #41: Flashing with 9600 baud works with Raspbian, but not on MacOS

tinue, 2015-05-24:
Because I don't have a USB2Serial adapter yet for 3.3V, I use my Arduino Due for this purpose. The Microcontroller is disabled (Reset shorted to Ground), so the Arduino Due acts as a pure USB2Serial adapter using an ATMEL 16U2 with firmware as the converter chip.

This setup works fine on a Raspberry Pi, and I could successfully flash the firmware (after I reduced the "Maximum block sized for RAM and Flash writes", as suggested elsewhere).

On the Mac, I only get "Failed to connect". When the ESP8266 is not in firmware mode, I can talk to is successfully with the Mac (using "screen" or the Arduino IDE).

After the firmware upgrade the ESP8266 switched from 9600 to 115200 baud. Now I can successfully flash also with the Mac.

All in all, it looks like the Mac can't work with 9600 Baud, but has no issues with 115200 Baud. The Raspberry Pi on the other hand can work with both speeds.

marcoskirsch, 2015-06-08:
I have successfully flashed ESP01 from OS X using a FT232R USB UART adapter I bought at Fry's.

projectgus, 2015-09-02:
I'm not sure there's anything we can do about this from the esptool end, unfortunately. It seems like a really unusual bug.

I tried reproducing the problem with a [Freetronics USBSerial](http://freetronics.com/usbserial) adapter, which runs basically the same atmega16u2-based firmware as the Due, but it worked OK at 9600bps for me. I'm on Linux though, so probably very similar driver to the RPi.

The default flashing baud rate for esptool.py is 115200, and I use the 16u2 USBSerial successfully at 230400 usually. So I guess the only reason to drop right down to 9600 would be if you didn't trust your signal integrity and wanted a really slow, stable signal.

So it could be an OS X driver issue with 9600bps, I guess? Or maybe a clock stability issue with the particular 16u2 you have, although I would expect that to be a bigger problem at the higher baud rates.

Please reopen the issue if you think there's something we can do from the esptool.py end to fix this.

closed 2015-09-02 by projectgus