espressif/esptool #43: elf2image error with sdk 1.1.0

tuanpmt, 2015-05-28:
```
make all 
../tools/esptool.py elf2image .output/eagle/release/image/eagle.app.v6.out -o ../bin/
Traceback (most recent call last):
  File "../tools/esptool.py", line 586, in <module>
    image.entrypoint = e.get_symbol_addr("call_user_start")
  File "../tools/esptool.py", line 355, in get_symbol_addr
    self._fetch_symbols()
  File "../tools/esptool.py", line 352, in _fetch_symbols
    self.symbols[fields[2]] = int(fields[0], 16)
ValueError: invalid literal for int() with base 16: 'U'
make[1]: *** [.output/eagle/release/bin/eagle.app.v6.bin] Error 1
make: *** [.subdirs] Error 2
```

themadinventor, 2015-05-28:
The call_user_start symbol is missing; either it has been renamed in the SDK, or it is missing due to an incorrect link. See also #24.

I'm thinking of adding a command line argument to specify the entry point name and/or adding a list of entry point names to look for.

tuanpmt, 2015-05-28:
this is my try (line 352)
```
try:            
    self.symbols[fields[2]] = int(fields[0], 16)
except ValueError:    
    print fields
```
and the output:
```
['U', 'os_intr_lock']
['U', 'os_intr_unlock']
``

cal101, 2015-06-05:
I found thats not necessarily the "call_user_start" symbols thats missing but 
sometimes missing symbols didn't make the linker fail.
You see that is the case here, too: os_intr_lock is missing here
Can't see a pattern when they get through the linker.
When doing an "nm" on the image you see the unresolved symbols.
In this case I guess I know the reason for the problem.
Unfortunately nodemcu stuff does not compile with -Wall and so it does not warn on missing prototypes but just guesses one. 
SDK 1.0.1 has defines for os_intr_lock etc. that map to (from memory) ets_intr_lock.
Those defines are gone for SDK 1.1.0 and so the linker looks for the functions os_intr_lock.
I added a compat header that has those defines.

projectgus, 2015-08-30:
I think the entry symbol name issue is fixed as of a0c18eac57f, so I'm going to close this issue. If you think I missed something then please reopen it.

closed 2015-08-30 by projectgus

tuanpmt, 2015-09-01:
It's working now, thanks