espressif/esptool #50: Chip id

sobiso, 2015-06-16:
Is there any chance to adding read chipid ?

projectgus, 2015-08-30:
I don't see why not. PRs would be welcome!

themadinventor, 2015-09-18:
What kind of chip id? esptool have been able of reading both the OTP MAC address and the SPI flash manufacturer id for some time :)

projectgus, 2015-09-18:
There's a 4 byte (from memory) unique(?) ID that isn't part of either MAC address, as far I know.

In esp-open-rtos we call it the OTP_CHIPID register (I can't remember how to access it from the Espressif SDK, but there's a way to do that as well):
https://github.com/SuperHouse/esp-open-rtos/blob/master/core/include/esp/dport_regs.h#L46

urmilparikh, 2016-02-20:
Added a pull request #85

projectgus, 2016-02-23:
This is closed by #85.

My comment about the OTP_CHIPID register @ 0x3ff00058 seems to be wrong, that address is some other internal chip identity information of some kind (qv http://esp8266-re.foogod.com/wiki/System_get_chip_id_%28IoT_RTOS_SDK_0.9.9%29 / http://esp8266-re.foogod.com/wiki/Get_chip_version_%28IoT_RTOS_SDK_0.9.9%29 ) but it's not an serial number type ID.

closed 2016-02-23 by projectgus