espressif/esptool #56: single file operation

eriksl, 2015-07-22:
Can you please add single file operation?

If the irom_text region starts at 0x10000, the gap from 0x8000 is quite small, I would like it...

projectgus, 2015-08-30:
Hi Erik,

By this do you mean creating an image for the SDK bootloader v1.2+ (or other bootloaders) where the irom section is included in the image?

eriksl, 2015-09-01:
Something like that.

I now do it explicitly, I take the iram section (the *0x00000* file), I pad it to 0x10000 and then append the irom section (*0x10000* file) to it. Not really rocket science ;-) It would be nice if esptool could do that itself. It may need an extra option to specify where the irom section starts.

projectgus, 2015-09-02:
I would find this useful as well. I'm going to merge this issue with #39 and put some more thoughts there, please feel free to continue discussion there. :)

closed 2015-09-02 by projectgus