espressif/esptool #58: uploading a SPIFFS partition corrupt existing firmware and/or wifi configs

martinayotte, 2015-08-17:
I have a firmware that is using SPIFS already uploaded. If I'm uploading a 64K SPIFFS partition at 0x6B000 using esptool.py, the firmware is not able to connect to Wifi wihout crashing.
If I upload the same partition with IGRR esptool, it is working fine.

Using esptool.py (crash at Wifi connection) : 
esptool.py --port /dev/ttyUSB0 write_flash 0x6B000 spiffs_part.bin

Using IGRR esptool (working fine) :  
esptool -vv -cd ck -cb 115200 -cp /dev/ttyUSB0 -ca 0x6B000 -cf spiffs_part.bin

I presume that esptool.py is touching/erasing some other parts of the flash that should not be accessed ...

projectgus, 2015-08-30:
Hi Martin,

Thanks for reporting this. I think it was fixed just now in #53, the longstanding SPI erase bug.

I'm going to close the issue but please reopen if you get the latest esptool.py and it's still broken.

Angus

closed 2015-08-30 by projectgus

martinayotte, 2015-08-30:
Yes, it seems to work now !
Thanks !