espressif/esptool #60: fail to download with multiple files

nqd, 2015-08-31:
Things go well with commit e96336f6561109e67afe03c0695d1e5b0de15da6.

When I update to current code, esptool fail to flash multiple files.

With 2 file, esptool goes well
```
../../tools/esptool/esptool.py --port /dev/ttyUSB0 write_flash -fs 32m-c1 0x00000 ../../esp-sdk/bin/boot_v1.4\(b1\).bin 0x01000 .firmwares/4096_user1.bin
Connecting...
Erasing flash...
Writing at 0x00000800... (100 %)
Erasing flash...
Writing at 0x0003f800... (100 %)

Leaving...
```
, but with 4 files
```
../../tools/esptool/esptool.py --port /dev/ttyUSB0 write_flash -fs 32m-c1 0x00000 ../../esp-sdk/bin/boot_v1.4\(b1\).bin 0x01000 .firmwares/4096_user1.bin 0x3FC000 ../../esp-sdk/bin/esp_init_data_default.bin 0x3FE000 ../../esp-sdk/bin/blank.bin
Connecting...
Erasing flash...
Writing at 0x00000800... (100 %)
Erasing flash...
Writing at 0x0003f800... (100 %)
Erasing flash...
Writing at 0x003fc000... (100 %)
Erasing flash...
Traceback (most recent call last):
  File "../../tools/esptool/esptool.py", line 579, in <module>
    esp.flash_begin(blocks*esp.ESP_FLASH_BLOCK, address)
  File "../../tools/esptool/esptool.py", line 214, in flash_begin
    raise Exception('Failed to enter Flash download mode')
Exception: Failed to enter Flash download mode
```.

Do you have any suggestion to overcome this?

Many thanks.

projectgus, 2015-08-31:
Hmm, it looks like the fix for the flash erase issue might have caused this regression. It's actually surprising flashing all four images worked with the old revision, most people have encountered the flash erase issue when flashing that much.

Can you please post the sizes (in bytes) of the four files you are flashing? That should make it easy to reproduce the exact problem, by creating dummy binary files.

nqd, 2015-08-31:
Sure, this is the 4 files https://www.dropbox.com/sh/8nszizvwose778r/AAD-yMjdXGPnBTh9SblIlXBma?dl=0

Thanks.

projectgus, 2015-09-02:
I think this should fix the underlying bug, please let me know if it doesn't.

Turned out it wasn't multiple files, it was that flashing the last file near the end of the flash exposed a bug.

closed 2015-09-02 by projectgus

nqd, 2015-09-02:
Great thanks. Will check next day when I touch the esp again.

nqd, 2015-09-03:
Confirmed. Thanks again!

commit 582a7cab (2015-09-02): Remove floats from block calculations, fix regression in 0e67846 #53

Float argument strayed into flash_begin() size, caused erase sizes to be
calculated as too long. This triggered an error when writing near the end of
the flash, it would ask for erasing beyond the end of available space.

Closes #60