espressif/esptool #63: Publish to PyPI.

skorokithakis, 2015-10-26:
Can you do a `python setup.py sdist upload` so esptool gets published to PyPI? Then we'd be able to install it with just a `pip install esptool`.

projectgus, 2015-10-26:
This sounds like a great idea.

I have never uploaded anything to PyPI before. Are you in a position to give any advice about whether esptool is ready for submission in its current state, or should other checks be done?

Angus

pfalcon, 2015-10-26:
What about this for example: https://github.com/SuperHouse/esp-open-rtos/blob/master/lib/symbols_norename.txt#L1 ?

projectgus, 2015-10-26:
> What about this for example: https://github.com/SuperHouse/esp-open-rtos/blob/master/lib/symbols_norename.txt#L1 ?

Hi @pfalcon,

Sorry, did you mean to leave this comment somewhere else? It's not clear to me how it relates to PyPI submission.

However, thanks for pointing that line in esp-open-rtos out. esptool.py no longer hardcodes the entry point name, as of some months ago it's loaded from the ELF file. In esp-open-rtos that `symbols_norename.txt` file is also no longer used and [should have been removed](https://github.com/SuperHouse/esp-open-rtos/commit/51c752dbcb754e77037b89f066b53b157b361991), all the symbols it lists are found in either our own source or newlib.

pfalcon, 2015-10-26:
Nope, I meant to leave it here - I believed that issue was fixed, but as I was found to mix up things, I'd let you to confirm it (before publishing to pypi without that patch possibly merged).

> should have been removed

Cool, one issue confirmed fixed, another is fixed. ;-)

skorokithakis, 2015-10-26:
What I would change before submitting to PyPI would be to make the script be called `esptool` after installation, rather than `esptool.py`, and to improve the documentation a bit. As a complete newbie to using it, I'm not very clear on how to use it. Then again, that might be more because I don't really know what it does or how, rather than a failure in the documentation.

projectgus, 2015-10-26:
Good suggestions, thanks @skorokithakis.

> Then again, that might be more because I don't really know what it does or how, rather than a failure in the documentation.

I think that's a failure in documentation, in either case. :)

skorokithakis, 2015-10-26:
Haha, possibly, but it may be out of scope for esptool.

pfalcon, 2015-10-26:
> would be to make the script be called esptool after installation, rather than esptool.py

Keep in mind that there's another "esptool". I believe it's not even maintained by the original author, but it was picked up by folks who like to ship binaries in peoples' faces. Then more community-concious of them call it "esptool-ck", but I'm not sure if everyone does. And then the best thing this project can do keep people from confusion is to continue being called esptool.py .

skorokithakis, 2015-10-26:
Sure, that's up to you. All I have is this esptool, but if you guys think that a collision has a reasonable chance of occurring, `esptool.py` is fine.

commit 00990b4b (2016-04-21): Update README to recommend pip installation

Closes #63
closed 2016-04-21 by projectgus