espressif/esptool #64: Properly reboot after reflash ?

kommancs96, 2015-11-07:
My configuration is the following:

ESP8266-01
Silabs USB to UART
Arduino IDE to compile code
esptool.py to upload the binary at 9600 baud rate.
Connections:
esp8266 USBtoUART
GND----------------------GND
VCC-----------------------3.3V
TX-------------------------RX
RX-------------------------TX (with voltage divider to 3.3v)
GPIO0--------------------GND
CH_PD-------------------3.3V

I upload my code to the esp8266 and it works fine, I can control an LED from my browser.
However , when I unplug the USB from my laptop and plug it in to a power supply or simply replug to my MacBook , it seems that the chip looses the firmware. After I unplug the chip I also free GPIO0 or connect to high with a resistor.

How do I properly reboot my chip after reflashing firmware?

projectgus, 2015-11-08:
Are you setting the baud rate in your code? During a post-reset execute, the default baud rate is the same as the uploading baud rate (so 9600 in your case). But after a hard reset it's the default 76800bps again, until you set it in your code.

Other than that and checking the pin connections are correct, running from a hard reset should be the same as running immediately after an upload.

If you connect your ESP-01 and set the baud rate to 76800 then you should see some bootloader messages before your firmware runs (or fails to run), as well.

projectgus, 2016-11-07:
Closing as it's been a year with no response, but feel free to reopen if you still think there's an issue here.

closed 2016-11-07 by projectgus