espressif/esptool #66: Not linked functions cause ValueError, not descriptive enough.

alex-march, 2015-11-09:
This has already been referenced in #43 
```
ValueError: invalid literal for int() with base 16: 'U'
```
If a function was defined but not linked and not called in code, the esptool.py, frightfully so picks it up as undefined and it's field[0] is set to 'U' instead of its address. Wouldn't it make sense to add a try/except and print something descriptive like "Probably forgot to link function " + field[1]? Around this line https://github.com/themadinventor/esptool/blob/master/esptool.py#L381. Since the python error message doesn't make much sense to an unfamiliar user.

projectgus, 2015-11-10:
Thanks for mentioning this, alex. I actually have a fix for this (and a few other errors) in a branch. I'll test it some more and commit ASAP. :)

closed 2015-11-10 by projectgus

alex-march, 2015-11-10:
Great! Thank you for such a quick reply!

commit 650d49f0 (2015-11-10): Add more informative errors for bad responses, missing symbol names

Closes #66