espressif/esptool #67: FTDI232R randomly stopping the upload

crushedice2000, 2015-11-17:
When I use a FTDI232R USB --> Serial IC, the connection breaks randomly (on my computer between 6-12% of the upload progress). I can fix this changing on `esptool.py` this

    ESP_RAM_BLOCK = 0x1800
    ESP_FLASH_BLOCK = 0x400 

by this

    ESP_RAM_BLOCK = 0x180
    ESP_FLASH_BLOCK = 0x40

Could you add a switch (like -s/--slow or -f/--ftdi) to make a slow transfer?

projectgus, 2015-11-17:
Hi crushedice,

I've seen some other people reporting that shrinking the block sizes works for them as well.

Have you tested with the very latest esptool.py? Some fixes got merged recently that might also help this situation.

If it still fails with the very latest, could please you also post an example full esptool.py command line and the full output from a failing command (with the default ESP_RAM_BLOCK/ESP_FLASH_BLOCK sizes)?

We can add a flag to use smaller transfer blocks, but I'd like to absolutely rule out any other kind of bug first.

Cheers,

Angus

crushedice2000, 2015-11-18:
Hi Angus,

I'm using the latest version (I've issued `git clone {awfullylongurl}` command yesterday).

I've read on the ESP forum that this may be because the FTDI skips packets, but I thing that it may be software-releated,

Ok, I'll send you all the material in a while.

Cheers,

Helio (formerly crushedice2000 ;-)

crushedice2000, 2015-11-18:
I'm really confused. Now it works with the default parameters. :confused:

I don't made any special thing. The only thing that changes is that now I'm flashing Basic instead of NodeMCU Lua

crushedice2000, 2015-11-18:
Huh!? It flashes well the esp8266basic firmware but stops at random percentage with NodeMCU. NodeMCU flashes well at low speed.

crushedice2000, 2015-11-18:
Here you're a asciinema recording of my session: https://asciinema.org/a/3tdlyhhbvqo996ospmc72hkwi

I attach here both firmwares in a zip file (renamed as TXT because GitHub doesn't allow me to upload ZIP)
[Zipped firmwares (please rename extension to zip).txt](https://github.com/themadinventor/esptool/files/37816/Zipped.firmwares.please.rename.extension.to.zip.txt)

crushedice2000, 2015-11-18:
If you need to have <s>straced :laughing:</s> ssh access to my computer to try things on the ESP8266, feel free to request it me.

projectgus, 2015-11-18:
Thanks for all this, especially appreciate the asciinema recording - I hadn't seen that before, very cool!

I would like to try to reproduce this... What kind of FT232R adapter board are you using, and where did you get it? Do you think it's more likely to be genuine or fake?

crushedice2000, 2015-11-19:
Asciinema is handy for these things ;-)

---

About the FTDI board, I'll try to answer all your questions:

> What kind of FT232R adapter board are you using?

A cheap one, bought in China (as all are made in China, even the ones I can buy in the local hardware store).

> Where did you get it?

http://www.aliexpress.com/item/FT232-Brush-USB-To-TTL-Serial-Cable-Line-Electric-Adapter-FTDI-Chipset-Computer-FT232RL/2026090019.html

---

> Do you think it's more likely to be genuine or fake?

I don't know how to test it. I can't try to "brick" it with the latest FTDI driver because I don't use Windows, but only OS X/Ubuntu.

However I think that it's fake, because the reference isn't laser etched, but printed with some white ink.

I've found on Google this picture to determine if the FTDI is original:

[image]
**Source:** http://hackaday.com/2014/02/19/ft232rl-real-or-fake/

---

I also attach some photos of my FTDI chip (I've removed the USB plastic shield and I've found a pretty small and smartly made board that allows me with a soldering iron to change the voltage between 5v, 3v3 and V<sub>IN</sub>)

[image]
[image]
[image]

---

What I consider a strange thing is that, with the default parameters (`0x400` and `0x1800`), the ESP8266Basic can flash correctly, but the NodeMCU Lua firmware stops. This remembers me to [this](http://www.catb.org/jargon/html/magic-story.html)

crushedice2000, 2015-11-19:
The asciinema was recorded on a late 2013 iMac with OS X 10.11, Now I've reproduced the problem on a Dell Vostro 3750 laptop with Ubuntu 14.04 with the same results

crushedice2000, 2015-11-19:
In Ubuntu it even fails with the ESP8266Basic firmware. Here youŕe the asciinema.

https://asciinema.org/a/9jp4dpcr02q7vfw2gke1b17a5

However this is another problem that should be in another issue.

crushedice2000, 2015-11-23:
@projectgus now arrived a NodeMCU (or something like from Ali) and it works correctly. The USB->Serial adapter is a CP2102. I believe that is a ESP8266-specific issue.

crushedice2000, 2015-12-26:
@themadinventor @projectgus: What can happen?

bobek, 2016-02-08:
I have very same issue. I am on Debian sid, tested with kernel 4.3.0 as well as 4.4.0. No difference. I have tried to flash using Windows machine and was not successful. My FTDI adapter is using the same chip as @crushedice2000 

Interestingly, whole flow was working last week. I am not aware of any real change in the setup.

Anyway :+1: to @crushedice2000 as I have been able to flash firmware after 2h of fiddling with caps and pullup resistors ;)

crushedice2000, 2016-02-09:
@bobek: This is really strange... Now I have 5 NodeMCU (one fried when I used a GPIO for input at 220V AC logic level). And they don't do anything bad. Flashing works fast. The chip that NoceMCU uses is CP2102. However, I can't imagine that the FTDI is "worse than others".

Is not good if more people has my problem. However, I'm happy to find a issue companion :smile_cat:
Maybe we can solve the issue.

Can be a problem of the mA that the FTDI can deliver for power? I don't know. Now I'm flashing with the slow method and connecting the ESP to Vᵢₙ (5V) and it's alive. :smile:. *Children, don't do this at home!* :smiling_imp:

projectgus, 2016-03-04:
> Can be a problem of the mA that the FTDI can deliver for power? I don't know

I missed this comment the first time around. **This is the problem**, FTDI chips' internal 3.3V regulator is not enough to power an ESP8266 - it will brown out during flashing but it will *really* brown out when you try to use the ESP8266 and the WiFi starts up. You need an external power supply.

The trick mentioned in the first post of using smaller blocks probably worked around the brownout by using power in smaller hits, but it's not a solution - just a bandaid.

I've heard elsewhere as well that powering with 5V doesn't immediately "let the magic smoke out", but if your ESP8266 goes pop later then you'll know why!

(I was coming here to post that I found an FT232R here, exact same markings as the photo @crushedice2000 posted - same date code, same product number... they're either genuine chips from the same batch,or fake chips that copied the same batch. It works every time.)

I'm going to close this issue. The front page README now stresses that power power stability problems can cause flashing to fail part-way through:
https://github.com/themadinventor/esptool#write_flash-operation-fails-part-way-through

Sorry it took us so long to resolve it with certainty!

closed 2016-03-04 by projectgus

crushedice2000, 2016-03-05:
@projectgus: Thanks!