espressif/esptool #68: esptool.py write the flash successfully, but nodemcu doesn't work or boot.

BG2BKK, 2015-11-20:
Hi! I compiled the nodemcu-firmware and got the 0x00000.bin and the 0x40000.bin. Then I typed "make flash" and writed done, then I turn the power off and on, but it didn't work. The same as I typed “sudo python esptool.py ..........”. When I switched to Windows 7 and used the https://github.com/nodemcu/nodemcu-flasher,  it worked. So what is the point I missed?

projectgus, 2015-11-20:
Hi!

What command line arguments did you use with esptool.py? And what module/device are you flashing?

The flash size & type arguments in particular may be important, depending on what kind of module you are flashing.

BG2BKK, 2015-11-20:
Hi!
Thank you for your reply.
My device is nodemcu DEVKIT 0.9, this one : http://nodemcu.com/index_en.html#fr_54747661d775ef1a3600009e.

I am newbie to eps8266 so that I get the the flash size & type by default. I type "make flash" under directory nodemcu-firmware. The cmd "../tools/esptool.py --port /dev/ttyUSB0 write_flash 0x00000 ../bin/0x00000.bin 0x10000 ../bin/0x10000.bin" is executed in fact.   The size of 0x00000.bin is 35KB(35024)  and the 0x10000.bin is 395kb(395588).

projectgus, 2015-11-21:
The default flash size & speed settings should work with that board.

The only other suggestion I have is to run an "esptool.py --port /dev/ttyUSB0 erase_flash" before the write_flash command you're using. Does that help?

projectgus, 2015-11-21:
If the erase_flash step doesn't work, can you paste the output of "esptool.py --port /dev/ttyUSB0 flash_id" please?

BG2BKK, 2015-11-21:
Hi! Here is the output of the cmd

$ sudo python ./esptool.py --port /dev/ttyUSB0 erase_flash
Connecting...
$ sudo python ./esptool.py --port /dev/ttyUSB0 write_flash 0x00000 ../nodemcu-firmware/bin/0x00000.bin 0x10000 ../nodemcu-firmware/bin/0x10000.bin 
Connecting...
Erasing flash...
Wrote 35840 bytes at 0x00000000 in 3.5 seconds (82.7 kbit/s)...
Erasing flash...
Wrote 396288 bytes at 0x00010000 in 38.4 seconds (82.5 kbit/s)...

Leaving...

It doesn't work again :D

projectgus, 2015-11-22:
Thanks for posting that. Really strange behaviour!

How about the other command I mentioned:

> If the erase_flash step doesn't work, can you paste the output of "esptool.py --port /dev/ttyUSB0 flash_id" please?

BG2BKK, 2015-11-23:
Thank you for your effort!
This is my output:
```lua
$ sudo ./esptool.py --port /dev/ttyUSB0 flash_id
Connecting...
Manufacturer: e0
Device: 4016

$ sudo ./esptool.py --port /dev/ttyUSB0 read_mac
Connecting...
MAC: 18:fe:34:00:d2:9a
```

It's weird that the nodemcu-flasher on Windows could work, using the same 0x00000.bin and 0x10000.bin. 
I can send you the two bin files by email.

iMihael, 2015-12-22:
Same behavior :(

asimzeeshan, 2015-12-23:
I had the same issue, I just changed my cable and flashed it. It worked
Something was wrong with my other cable although it was charging well

iMihael, 2015-12-23:
I resolve the issue. I'm change baud rate for esptool to 9600. And It work!
`./esptool.py --baud 9600  write_flash 0x00000 ../nodemcu-firmware/bin/0x00000.bin 0x10000 ../nodemcu-firmware/bin/0x10000.bin`

iMihael, 2015-12-23:
And after flashing nechesarili plug out and plug in the cable.

BG2BKK, 2015-12-24:
Hi~
Here is my current station:
I download the bin-file and src-file of v0.9.6 firmware  frome https://github.com/nodemcu/nodemcu-firmware/releases/. I burnt the bin-file with esptool as the  https://github.com/themadinventor/esptool writes, and it succeed to enter the lua environment after restart. Now the problem is, when I compile the nodemcu-firmware and get 0x00000.bin and 0x10000.bin, I succeed to burn them, but the nodemcu doesn't work. So what next?

BG2BKK, 2015-12-24:
Hi guys~
Maybe we all ignore something important such as boot mode or flash mode of esp8266, for example https://zoetrope.io/tech-blog/esp8266-bootloader-modes-and-gpio-state-startup. @projectgus Could you teach us something about the boot mode :D

iMihael, 2015-12-24:
As I understood if you have nodemcu development kit, you don't need to worry about this.

BG2BKK, 2015-12-26:
Hi guys~
Maybe I solved this issue. Here is my solution

```bash
 sudo esptool.py --port /dev/ttyUSB0 write_flash -fm dio -fs 32m -ff 40m 0x00000 0x00000.bin 0x10000 0x10000.bin
```
It is clearly confirmed that args such as -fm, -fs and -ff  should be specified. 
If it still doesn't work, try to do sth before you burn your bin-file:
1. press the **FLASH** button (don't release)
2. press the **RST** button and release
3. release **FLASH** button

Then it should work correctly.
good luck~

marcelstoer, 2016-02-07:
> ...args such as -fm, -fs and -ff should be specified

What a pity that SPI flash info args are not documented anywhere here (I expected that in `README.md`).

projectgus, 2016-03-04:
Closing as `-fm dio` was the missing parameter here, and this is now documented in the README.

closed 2016-03-04 by projectgus