espressif/esptool #69: Significance of 'ELF binary has undefined symbol...'

raburton, 2015-11-23:
I'm getting `A fatal error occurred: ELF binary has undefined symbol __errno` since moving to SDK v1.4 and I don't really know if it's important. It stops esptool working, but can easily be worked around. I can't understand how the linker is happy if there are undefined symbols.

projectgus, 2015-11-24:
I can't really explain that either off the top of my head.

Are you able to send me an example ELF (and ideally a .map file as well), please? Can attach it to the Issue or email it to me privately. OK if not.

Angus

raburton, 2015-11-26:
I've put the files on dropbox, it won't let me attach them here. These were built with Sming (c++), but the error only appeared when I moved to sdk v1.4. I had to update esptool.py as well, initially with sdk v1.4 it couldn't handle the U type symbol, after updating it handled it (by throwing an error). I assume this means other people have started to get these undefined symbol errors too, since there was a recent change to handle them.

https://dl.dropboxusercontent.com/u/5500141/app.a
https://dl.dropboxusercontent.com/u/5500141/app.map

zardam, 2015-11-26:
This is not related to esptool.py, see SmingHub/Sming#450 In fact, I think esptool.py is handling the situation correctly by throwing an error.

raburton, 2015-11-27:
> This is not related to esptool.py, see SmingHub/Sming#450 In fact, I think esptool.py is handling the situation correctly by throwing an error.

That isn't the question, the question is what is the significance of this. Undefined symbols should be an error at the linker stage, so what does it mean to get through the linker fine and only complain at esptool.

zardam, 2015-11-27:
It seems to be related to the --gc-sections linker flag. Without it, I have a linker error.

projectgus, 2015-11-30:
@raburton, can you post the linked .elf file as well please?

I think what's probably happening is Sming is compiling with -ffunction-sections or -fdata-sections, but then linking without --gc-sections. This means there are unused symbols in the ELF file, brought in by other symbols in the same source file. These are what causes esptool.py to fail, as usually this indicates an unlinked ELF file. However those unused symbols also happen to be in unused sections so it doesn't really matter and objcopy can still extract a binary.

The "proper" fix is probably to link with --gc-sections. But we should handle this case properly in esptool.py as well.

That's all a guess at this stage, though. It'd be good to confirm it from the ELF file if possible.

raburton, 2015-12-02:
Sorry, should have included that last time.
https://dl.dropboxusercontent.com/u/5500141/app.out
It's been rebuilt in the mean time, so in case it's critical that the others files (linked above) match for what you want to look at I've updated them too.

Sming is linking with --gc-sections already.

closed 2015-12-03 by projectgus

projectgus, 2015-12-03:
Thanks for sending that, very helpful.

Short answer: Looks like you can have undefined symbols in fully linked binaries, as long as they never get referenced. So I've downgraded that fatal error to a warning.

Long answer:

@zardam's diagnosis in SmingHub/Sming#450 seems to be correct, and I probably should have looked at that Issue before spelunking in the ELF file myself! After going over the build steps myself, I don't quite see how Sming ever uses the esp_microc header files based on the default install steps for Linux - they don't show up anywhere in the source tree. I guess there must be some step I missed.

It's not clear to me why this seemed to break when upgrading to Espressif SDK v1.4.0. None of the Espressif SDKs ship an errno.h header. Or did you mean Sming v1.4.0?

Anyhow, although it's really a bug in the compiled source code I think it makes sense for esptool.py to do the best it can with the ELF it gets - if the linker succeeded then all of the symbols actually needed in the binary are there, and that's all that matters in the end.

OTOH, if the binary turns out to have actually invalid contents then esptool.py will now print a warning, and then fail outright when objcopy fails to extract the text sections.

raburton, 2015-12-03:
Thank you for looking at this.

Richard.

commit 50459db2 (2015-12-03): elf2image: Downgrade missing symbols to a warning not an error

Closes #69. Situation where an unused reference to an undefined symbol
make it into a fully linked ELF.

If a missing symbol is essential (ie the ELF really is on partially linked) then the
objcopy step will still fail so esptool.py will fail.