espressif/esptool #71: Corruption when writing to 25Q80BSIG flash using QIO

dreamlayers, 2015-12-09:
I have a module with GigaDevice 25Q80BSIG 8mbit flash. Esptool uses QIO by default, and that leads to corruption when writing flash. Most of the chip gets written properly, but some data ends up in the wrong place, as if some address bits get flipped, commonly in some specific small areas of the chip. It does not seem like data bits get flipped. Reads seem fine with QIO. Going to 20 MHz SPI clock does not help. Adding a 10 microfarad tantalum capacitor at the module also didn't help. I don't get any corruption with DIO.

The original firmware was set up for QIO, and it worked. I modified esptool to write firmware set up for QIO even when writing using DIO, and the new firmware also works with QIO. Of course the firmware does a lot less writing during normal operation, so I can't be sure it won't corrupt the flash sometimes. I realize this could be a hardware issue with the module.

projectgus, 2015-12-15:
Sorry for the slow reply.

This is a really strange problem! I think if the checksum that the serial bootloader calculates matches what esptool expects, and the serial protocol otherwise succeeds, then I'm not sure if there's anything we can do from the esptool end.

Do you have multiple modules that all display the same problem? I guess otherwise my money would be on faulty SPI flash before anything else.

Angus

projectgus, 2015-12-29:
I'm going to close this issue now, but please reopen if you think esptool.py is doing something wrong.

closed 2015-12-29 by projectgus