espressif/esptool #73: provide --tmpdir that exposes the dir parameter of mktemp

m-bruck, 2015-12-18:
When running esptool on a cygwin python combined with a non-cygwin objcopy the temp file is placed in virtual paths that are invisible to non-cygwin programs (`/tmp`).

Unfortunately specifying a local `TMP`/`TEMP`/`TEMPDIR` does not resolve the issue as the path is then replaced with an absolute path inside the virtual cygwin filesystem root again (`/cygdrive`).

However specifying a local path in mktemp works fine:
`tmpsection = tempfile.mktemp(suffix = ".section", dir = "my_local_temp_dir")`

So a solution would be to make the `dir` parameter available via the command line.

projectgus, 2015-12-29:
Thanks for reporting this.

I'm not against exposing the tmpdir option, but I'd also like to see if we can fix the problem so it "just works" for cygwin users - fixing the problem via the command line option means all cygwin users will have to know that this is a problem, then find the option, then choose a suitable local/relative path.

It looks like [the only foolproof way to translate cygwin paths to Windows this is to run the cygpath command line tool](http://stackoverflow.com/questions/10884268/convert-posix-win-path-in-cygwin-python-w-o-calling-cygpath), but I think that might be a better long term solution (change cygwin-specific paths into Windows paths transparently, when we run objcopy).

What do you think?

projectgus, 2016-01-05:
@m-bruck, can you please try checking out the `fix/cygwin` branch on your system and let me know if it fixes the problem? It should ensure any path that goes to binutils is a Windows native path not a Cygwin path (which should work even if the toolchain is compiled against Cygwin, as Cygwin knows how to open either kind of path.)

m-bruck, 2016-01-05:
cygpath emits a newline so after adding .rstrip it works:

`return subprocess.check_output(["cygpath", "-w", p]).rstrip('\n')`

I agree that `--tmpdir` would have been more of a hack. I had issues with windows paths on cygwin tools in the past, I think with `make`, so I try to give tools their native paths. But a quick test shows that cygwin's native binutils and gcc seem to take all versions of `cygpath`'s output.

projectgus, 2016-01-07:
Thanks! I amended the commit with the stripped newline. Let me know if that doesn't fix the problem somehow. :)

commit 333db731 (2016-01-07): Call cygpath to sanitise paths for Cygwin Python

Fixes #73
closed 2016-01-07 by projectgus