espressif/esptool #77: Problem to write a spi file system at 0xeb000

wellrats, 2016-01-02:
I'm trying to use esptool.py to write a file system generated by mkspiffs at 0xeb000.

To generate the file system i use:

`````
mkspiffs -c html/bin webpages.espfs
`````

The address that SPIFFS is looking for:

````
SPIFFSImpl: allocating 512+180+1400=2092 bytes
SPIFFSImpl: mounting fs @eb000, size=10000, block=1000, page=100
SPIFFSImpl: mount rc=0
````

To upload the file system i use:

````
esptool.py --port=$(ls /dev/cu.usbserial* | head -1) --baud=921600 write_flash 0xeb000 webpages.espfs 
`````

esptool.py writes OK, but the system doesn't boot anymore so everytime i have to upload the sketch.
If i comment the line  ````flash_begin(0,0)```` at the end of write_flash command it writes OK and i don't have to upload the sketch again.

Is there a way or parameter for writing an SPIFFS without re-upload the sketch or comment this line ?

projectgus, 2016-01-08:
Is it possible you have 'dio' (as opposed to 'qio') flash? If you add `-fm dio` to the command line options, does it start correctly without needing a re-upload?

wellrats, 2016-01-08:
Yes, with ````-fm dio```` it worked.
Thanks !!!
Closing the issue.

closed 2016-01-08 by wellrats

projectgus, 2016-01-09:
Glad to hear!

Just out of curiosity, what's the output of 'esptool.py flash_id' on this board? And what kind of device is it?

Cheers,

Angus

wellrats, 2016-01-12:
Manufacturer: e0
Device: 4016
I don't know what happened but it's working now even if i don't use ```-fm dio```.

projectgus, 2016-01-13:
Thanks for the update. That's really odd! Glad it's working now.

kommancs96, 2017-02-18:
I'm having a similar problem. First how do I download `mkspiffs` ?
Then how do I find out the correct address to flash the filesystem next to the firmware?

projectgus, 2017-02-27:
mkspiffs is usually included as part of the software framework that you're using to program the ESP8266. Consult their documentation find out how they integrate SPIFFS, and how to use those tools.