espressif/esptool #79: Boot error after flashing on linux

gozotto, 2016-01-06:
I try to flash the esp-link firmware with the esptool and got a error during boot.
See https://github.com/jeelabs/esp-link/issues/94
The same firmware flashed with NodeMCU-Flasher and espressif flash download tool (only user1.bin) works.
I've work mostly with linux and would prefer the usage of esptool.py. What's the difference in flashing with the flash tools?

gozotto, 2016-01-06:
Some console logs from both tools:
__espressif flash tool:__

get flash id : 0x401640c8
 manufacturer_id: 0xc8
 device_id: 0x4016
vendor: 200
mode: 64
size: 22
filename:  D:\Data\Projects\ESP8266\Tools\FLASH_DOWNLOAD_TOOLS_v2.3_150916\bin_tmp\downloadPanel1\user1.bin
offset :  4096
Erasing flash...
head:  15 ;total: 74
erase size :  241664

Writing at 0x00001000... (0 %) 
Writing at 0x00001400... (0 %) 
Writing at 0x00001800... (0 %) 
Writing at 0x00001c00... (1 %) 
Writing at 0x00002000... (1 %) 
...
Writing at 0x00049400... (98 %) 
Writing at 0x00049800... (98 %) 
Writing at 0x00049c00... (99 %) 
Writing at 0x0004a000... (99 %) 

__esptool.py:__
Writing at 0x00010000... (0 %) 
Writing at 0x00010400... (0 %) 
Writing at 0x00010800... (1 %) 
Writing at 0x00010c00... (1 %)  
...
Writing at 0x00058000... (98 %) 
Writing at 0x00058400... (98 %) 
Writing at 0x00058800... (99 %) 
Writing at 0x00058c00... (99 %) 
Writing at 0x00059000... (100 %)  

So the esptool.py writes more flash than espressif tool.

projectgus, 2016-01-06:
I saw on the parent issue that someone commented that the user1.bin offset is 0x1000 not 0x10000. I think this accounts for the different write address ranges in the output as well.

Can you confirm that changing the user1.bin offset fixes the problem?

gozotto, 2016-01-07:
Thank's a lot... that's the missing link.

closed 2016-01-07 by gozotto