espressif/esptool #82: esptool via Arduino

shayo, 2016-02-06:
Hi,

I'm having trouble flashing the 8266 via serial relay.
Here is the scenario:
esptool works great if I use FTDI. However, In my application, I need to flash it using a Mega32U4.
I hooked up the Mega32U4 => ESP like this:
3.3V => VCC
GND => GND
TX => RX
RX => TX

I then upload to the mega32U4 a serial relay program:
`void setup() {
   Serial.begin(115200);
  Serial1.begin(115200);
}

void loop() {
 while(Serial1.available())
    Serial.write((uint8_t)Serial1.read());

while(Serial.available())
      Serial1.write((uint8_t)Serial.read());
}
`

Then, I try to upload (all 8266 pins are set correctly, just like it worked with the FTDI), but I keep getting this error:

`Connecting...

A fatal error occurred: Failed to connect to ESP8266
`

Any idea why this is happening? I tried lower baud rate (9600) without any success.
This should work in theory....right?

shayo, 2016-02-06:
FYI, I'm not the only person who encountered this:
http://secretsciencelab.com/how-to-flash-espduino-firmware-on-cactus-micro-esp8266-rev1/

projectgus, 2016-02-06:
I agree this should be able to work.

Does it change anything if you replace the two while statements in the relay code with if statements?

shayo, 2016-02-06:
No. It doesn't matter. It also doesn't matter if I switch their position (i.e., trying to read from Serial first).
How can we debug this further?

chrismerck, 2016-02-07:
I also have this issue. If I manually run a python interpreter and write/read from serial, I can read back the response from the esp. However, when using esptool, we get "Invalid head of packet" exception from receive_response function. (The exception is caught and ultimately manifests as the "failed to connect" error.)

What is odd is, the self._port.read(1) call in receive_response is returning immediately, rather than waiting for the timeout of 300ms. So it appears to be a serial configuration issue.

chrismerck, 2016-02-07:
This worked for me:

Comment-out the "issue reset-to-bootloader" commands. Apparently this puts the serial port in a bad state:

```
162         for _ in xrange(4):
163             # issue reset-to-bootloader:
164             # RTS = either CH_PD or nRESET (both active low = chip in reset)
165             # DTR = GPIO0 (active low = boot to flasher)
166             """self._port.setDTR(False)
167             self._port.setRTS(True)
168             time.sleep(0.05)
169             self._port.setDTR(True)
170             self._port.setRTS(False)
171             time.sleep(0.05)
172             self._port.setDTR(False)"""
```

shayo, 2016-02-07:
Works!
Thank you Chrismerck!

It might be a good idea to add this as a flag that can be controlled from the command line arguments (or add this to the documentation somewhere).

projectgus, 2016-03-03:
Good suggestion, we should have a flag controlling the desired reset behaviour.

projectgus, 2017-01-10:
esptool.py v2.0 (current master branch) has a `--before no_reset` option which will keep DTR/RTS at default values and should fix this problem when using Arduino boards as serial interfaces.

Please reopen this issue if you find this isn't the case.

closed 2017-01-10 by projectgus