espressif/esptool #96: (much) faster flashing, better verification

rojer, 2016-03-22:
hi guys

TL;DR: i wrote an elaborate RAM stub that implements faster flashing and better verification.
i won't repeat the explanation here, 
please read the [blog post](https://blog.cesanta.com/esp8266-superfast-flashing-flash-esp-in-5-seconds) and let me know what you think.
interested / not interested?

projectgus, 2016-03-22:
I am very interested! Very nice work. :) Would be very interested to have this in esptool.py :)

I had been thinking some vague ideas along the same lines, also possibly compressing the payload in some way.

Are you thinking of sending a PR? (It would be very welcome!)

rojer, 2016-03-22:
yeah, why not. some additional clean up may be needed, but otherwise write_flash is already done - https://github.com/cesanta/fnc/blob/master/common/platforms/esp8266/stubs/esptool.py#L819

verify can be re-implemented to use on-device digest, that makes sense.

would you like to expose all of {wrap,run}_stub or take just enough to run it?

projectgus, 2016-04-04:
Hi rojer,

Are you able to relicense your C flasher stub code as GPL, please? The stub source files are marked Copyright Cesanta "All Rights Reserved", which makes them incompatible with GPL (and esptool.py is GPL licensed).

> would you like to expose all of {wrap,run}_stub or take just enough to run it?

I'm guessing the wrap_stub/run_stub functions were added to aid with stub development and integrate with FNC? Do I have that right?

I'm more inclined to just add a flasher stub, rather than add a generic stub framework. But maybe there are uses for those commands that I haven't thought of?

rojer, 2016-04-04:
as stated in the [https://github.com/cesanta/fnc/blob/master/LICENSE](top-level LICENSE file), all the code is already licensed under GPL.

yes, wrap/run_stub commands are there to aid stub development. ok, i'll take just the necessary part of run_stub.

projectgus, 2016-04-20:
Closing in favour of #103.

closed 2016-04-20 by projectgus