espressif/esptool #98: Flashing HSPI Flash

sskylark, 2016-03-27:
Can you see what happens when you do this after sync:
```
self.write_reg(0x3FF00028, 2, 2)
self.write_reg(0x60000800, 0, 0x300)
self.write_reg(0x60000804, 0x20, 0x130)
self.write_reg(0x60000808, 0x20, 0x130)
self.write_reg(0x6000080C, 0x20, 0x130)
self.write_reg(0x60000810, 0x20, 0x130)
```
This should set up DIO SPI0 on HSPI port. If someone has a board with HSPI Flash, I'd be curious to hear about the results. Note: make sure nothing external is driving the SDIO bus - the bootloader SPI code will try to enable it as SPI as well, so multidriver issues may occur.

At least get_flash_id seems to work for me now on HSPI :-) I have not actually tried flashing anything yet... (my code runs deeply embedded so I can't just run esptool.)

projectgus, 2016-03-28:
Neat!

So, just to make sure I understand what you did here - this is putting a DIO SPI flash chip on the HSPI bus and then using the "swap" feature in the ESP hardware to configure that as the primary SPI flash?

Angus

sskylark, 2016-03-28:
You got it right. The other accesses just set up HSPI on IOMUX and control SPI clocks.

Also, flashing does work with no other changes, and I can boot firmware (though this needs a few tweaks... by default, AT firmware will assign some of the HSPI pins to UART0).

st0ff3r, 2016-06-03:
yes, just tested - it works. Would it be possible with an option to do it?

st0ff3r, 2016-10-21:
at least it worked for NAND with 24 bit address, but not for my FeRAM with only 16 bits for address :-/

projectgus, 2017-06-14:
esptool.py v2.0 now [supports a command line argument](https://github.com/espressif/esptool/wiki/Advanced-Options#overriding-spi-flash-connections) to configure the SPI connection parameters on ESP32. The `--spi-connection HSPI` option does not work on ESP8266, but it could probably be made to work by making register writes as shown above.

gavkeeley, 2020-09-21:
Hi @projectgus, @sskylark or @st0ff3r would you mind explaining how/where to actually perform these 'self.write_reg()' commands? I've got a couple of ESP8266 ESP-WROOM-S2 boards which have their 2MB flash connected to the HSPI port and would appreciate any support. Thanks in advance.

The modules if you care to look: [https://za.rs-online.com/web/p/wlan-modules/1359774/](https://za.rs-online.com/web/p/wlan-modules/1359774/)