espressif/esptool #100: append_checksum error

rzajac, 2016-04-03:
Today I got this weird error 

```
Traceback (most recent call last):
  File "/opt/esptool-py/esptool.py", line 1019, in <module>
    main()
  File "/opt/esptool-py/esptool.py", line 992, in main
    operation_func(args)
  File "/opt/esptool-py/esptool.py", line 783, in elf2image
    image.save(args.output + "0x00000.bin")
  File "/opt/esptool-py/esptool.py", line 464, in save
    self.append_checksum(f, checksum)
  File "/opt/esptool-py/esptool.py", line 432, in append_checksum
    f.write(struct.pack('B', checksum))
struct.error: cannot convert argument to integer
```
It was caused by code below:

```
bool ICACHE_FLASH_ATTR bee_util_ensure_opmode(uint8 opmode, bool restart)
{
  bool success = true;

  if (wifi_get_opmode() != opmode) {
    LOG("changing opmode %d -> %d\r\n", wifi_get_opmode(), opmode);
    ETS_INTR_LOCK();
    success = wifi_set_opmode(opmode);
    ETS_INTR_UNLOCK();
    if (restart == true) {
      system_restart();
    }
  }

  return success;
}
```

but if I change it to:

```
bool ICACHE_FLASH_ATTR bee_util_ensure_opmode(uint8 opmode, bool restart)
{
  bool success = true;

  if (wifi_get_opmode() != opmode) {
    LOG("changing opmode %d -> %d\r\n", wifi_get_opmode(), opmode);
    ETS_INTR_LOCK();
    success = wifi_set_opmode(opmode);
    ETS_INTR_UNLOCK();
    if (restart == true) {
      system_restart();
    }
  }

  return true; // <--- here is the change
}
```

Everything works fine.

projectgus, 2016-04-03:
That *is* a weird error, especially given the fix!

Could you please let me see an ELF file that triggers the crash? You can attach it here, or if you prefer to keep it between us then you can email me - gus at projectgus dot com.

It's OK if you can't share the file.

Angus

rzajac, 2016-04-04:
Hey,

I have sent it to your email.

projectgus, 2016-04-06:
Thanks for sending that through. This was a silly bug where a checksum of zero was being accidentally interpreted as "no checksum" so the calculation was failing. Explains why changing absolutely anything in the ELF file would have fixed it (ie the checksum changed to something other than zero.) Should be fixed now.

commit c7eb51c2 (2016-04-06): Fix bug where checksum==0 (& other usage of missing 'is None')

Closes #100
closed 2016-04-06 by projectgus