espressif/esptool #102: "Invalid argument" for baud beyond 230'400

marcelstoer, 2016-04-20:
This is a follow-up to #101.

On OS X flashing a WeMos D1 mini which uses a CH340 adapter from Winchiphead fails for baud rates beyond 230'400. esptool reports "Invalid argument" in `set_special_baudrate`.

```
/Applications/esptool (master) > esptool.py --baud 921760 --port /dev/tty.wchusbserial1420 write_flash -fm dio -fs 32m -ff 40m 0x00000 ~/Downloads/nodemcu.bin 
Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 5, in <module>
    pkg_resources.run_script('esptool==0.1.0', 'esptool.py')
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 492, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 1357, in run_script
    exec_(script_code, namespace, namespace)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 47, in exec_
    exec("""exec code in globs, locs""")
  File "<string>", line 1, in <module>
  File "/Library/Python/2.7/site-packages/esptool-0.1.0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 724, in <module>
    
  File "/Library/Python/2.7/site-packages/esptool-0.1.0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 575, in main
    
  File "/Library/Python/2.7/site-packages/esptool-0.1.0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 71, in __init__
    
  File "build/bdist.macosx-10.9-intel/egg/serial/serialutil.py", line 345, in setBaudrate
  File "build/bdist.macosx-10.9-intel/egg/serial/serialposix.py", line 431, in _reconfigurePort
  File "build/bdist.macosx-10.9-intel/egg/serial/serialposix.py", line 148, in set_special_baudrate
IOError: [Errno 22] Invalid argument
```

gecgooden, 2016-09-11:
I'm having the same problem. Did you work out a solution to this?

marcelstoer, 2016-09-11:
No, I live with it.

mcauser, 2016-11-09:
I believe it is a problem with the OS X CH340G driver.
Baud 230400 and 115200 work for me on Yosemite and El Capitan.
Anything faster does not work, eg. 460800, 921600.

http://kig.re/2014/12/31/how-to-use-arduino-nano-mini-pro-with-CH340G-on-mac-osx-yosemite.html

Signed CH340G driver can be found here: https://www.wemos.cc/downloads

The new WeMos D1 Mini Pro replaced the CH340G with a CP2104.
https://www.wemos.cc/product/d1-mini-pro.html
Driver linked at the bottom of the page (windows, mac, linux).

commit 0228f57a (2017-06-14): Display user-friendly error if baud rate unsupported by driver

Closes #102
closed 2017-06-14 by projectgus