espressif/esptool #107: flashed but not booting

oczkers, 2016-05-07:
Hi guys, my problem is probably not related to esptool but i can't find anyone to help me.

I've got esp-12e which is flashing successfully but cannot load boot (tested with few versions of official ai thinker sdk, nodemcu, scripts uploaded via arduino ide etc.). Do You have any idea what could be the reason?

[flash & boot process in one image](https://i.imgur.com/ZpvQLIo.png)

projectgus, 2016-05-08:
Hi oczkers,

That's very odd, the bootloader stub is flashed and running but it can't run the user code for some reason.

Most ESP-12Es have 32mbit flash, do you get different results if you pass that as the size? (I can't think of a reason why 8mbit wouldn't work as well, but I figured it's worth a try.)

Does it work if you use Espressif's official flash download tool?

Angus

oczkers, 2016-05-08:
I doesn't meter what soft or settings (tried 32mb, 8mb, old firmware that fits 4mb) i use to flash, efect is always the same.
Is it possible that i have made a typo in address and overwrote some restricted area or damaged chip somehow?

projectgus, 2016-05-09:
It isn't possible to damage the ESP via the software bootloader, once you erase the chip and re-flash it should be as good as new.

It could be a hardware problem, either something that's connected to the ESP or maybe you have a defective ESP-12E.

Please try the other items mentioned here:
https://github.com/themadinventor/esptool#write_flash-succeeds-but-esp8266-doesnt-run

(I just added a new item there mentioning GPIO connections for the SPI flash, although I don't think that's your particular problem.)

Another thing to try is to download the stable 1.0.1 release and try it. You can get it here:
https://github.com/themadinventor/esptool/releases/tag/v1.0.1

projectgus, 2016-07-14:
Did you have any luck with this?

projectgus, 2016-07-22:
I'm going to close this, but feel free to reopen if the problem isn't gone.

closed 2016-07-22 by projectgus

oczkers, 2016-07-22:
Sorry for delay. I tried stable esptool but effect was exactly the same so i gave up, there is no need to reopen because it's not related to Your great soft.