espressif/esptool #109: sandbox violation on master

l29ah, 2016-05-17:
/usr/bin/python2.7 setup.py build

Error:
  * In program /usr/bin/cave --colour yes perform install --hooks --managed-output --output-exclusivity with-others =dev-embedded/esptool-9999:0::booboo --destination installed --replacing =dev-embedded/esptool-9999:0::installed --x-of-y 1 of 1:
  * When installing 'dev-embedded/esptool-9999:0::booboo' replacing { 'dev-embedded/esptool-9999:0::installed' }:
  * When running an ebuild command on 'dev-embedded/esptool-9999:0::booboo':
  * Install failed for 'dev-embedded/esptool-9999:0::booboo' (paludis::ActionFailedError)

Traceback (most recent call last):
  File "setup.py", line 90, in <module>
    'esptool.py',
  File "/usr/lib64/python2.7/distutils/core.py", line 111, in setup
    _setup_distribution = dist = klass(attrs)
  File "/usr/lib64/python2.7/site-packages/setuptools/dist.py", line 269, in __init__
    self.fetch_build_eggs(attrs['setup_requires'])
  File "/usr/lib64/python2.7/site-packages/setuptools/dist.py", line 313, in fetch_build_eggs
    replace_conflicting=True,
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 826, in resolve
    dist = best[req.key] = env.best_match(req, ws, installer)
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 1071, in best_match
    return self.obtain(req, installer)
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 1083, in obtain
    return installer(requirement)
  File "/usr/lib64/python2.7/site-packages/setuptools/dist.py", line 380, in fetch_build_egg
    return cmd.easy_install(req)
  File "/usr/lib64/python2.7/site-packages/setuptools/command/easy_install.py", line 640, in easy_install
    return self.install_item(spec, dist.location, tmpdir, deps)
  File "/usr/lib64/python2.7/site-packages/setuptools/command/easy_install.py", line 670, in install_item
    dists = self.install_eggs(spec, download, tmpdir)
  File "/usr/lib64/python2.7/site-packages/setuptools/command/easy_install.py", line 850, in install_eggs
    return self.build_and_install(setup_script, setup_base)
  File "/usr/lib64/python2.7/site-packages/setuptools/command/easy_install.py", line 1078, in build_and_install
    self.run_setup(setup_script, setup_base, args)
  File "/usr/lib64/python2.7/site-packages/setuptools/command/easy_install.py", line 1066, in run_setup
    raise DistutilsError("Setup script exited with %s" % (v.args[0],))
distutils.errors.DistutilsError: Setup script exited with error: SandboxViolation: mkdir('/var/tmp/paludis/dev-embedded-esptool-9999/work/esptool-9999-python2_7/flake8.egg-info', 511) {}

The package setup script has attempted to modify files on your system
that are not within the EasyInstall build area, and has been aborted.

This package cannot be safely installed by EasyInstall, and may not
support alternate installation locations even if you run its setup
script by hand.  Please inform the package's author and the EasyInstall
maintainers to find out if a fix or workaround is available.

projectgus, 2016-05-19:
I'm not sure exactly why you're getting the SandboxViolation here. It looks like it happens while setuptools is trying to install flake8, which is listed under setup_requires (so it's installed as part of the setup.py build process, but not included in the final package.)

I can't reproduce this, though. I just ran these commands to create a clean virtualenv (no pre-existing packages) and install esptool.py into it:

```
virtualenv2 ./testenv
source testenv/bin/activate
git clone https://github.com/themadinventor/esptool.git
cd esptool
python2.7 setup.py build
```

... completes as expected.

It looks like you're running a Gentoo ebuild environment? Any other clues you can provide?

l29ah, 2016-05-19:
I don't really know how setuptools work. If you have a gentoo nearby, you can try installing dev-embedded/esptool::booboo; it used to work some time ago. Otherwise i can bisect it if it helps.

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

Quarky9, 2016-05-26:
Same here on Gentoo:
setup.py build
errors out but builds up massive memory footprint and CPU load while doing so...
  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                      25927 portage   30  10 18.158g 0.017t   8332 R 100.0 55.4   0:35.09 python2.7              

 USE:        abi_x86_64 amd64 elibc_glibc kernel_linux python_targets_python2_7 userland_GNU
 FEATURES:   preserve-libs sandbox userpriv usersandbox
>>> Unpacking source...
 Fetching https://github.com/themadinventor/esptool ...
git fetch https://github.com/themadinventor/esptool +HEAD:refs/git-r3/HEAD
git symbolic-ref refs/git-r3/dev-embedded/esptool/0/__main__ refs/heads/master
 Checking out https://github.com/themadinventor/esptool to /tmp/portage/dev-embedded/esptool-9999/work/esptool-9999 ...
git checkout --quiet master
GIT update -->
   repository:               https://github.com/themadinventor/esptool
   at the commit:            56224490d8e37875a893a7c4ebb0f9a59b1e5548
>>> Source unpacked in /tmp/portage/dev-embedded/esptool-9999/work
>>> Preparing source in /tmp/portage/dev-embedded/esptool-9999/work/esptool-9999 ...
>>> Source prepared.
>>> Configuring source in /tmp/portage/dev-embedded/esptool-9999/work/esptool-9999 ...
>>> Source configured.
>>> Compiling source in /tmp/portage/dev-embedded/esptool-9999/work/esptool-9999 ...
 python2_7: running distutils-r1_run_phase distutils-r1_python_compile
/usr/bin/python2.7 setup.py build
Traceback (most recent call last):
  File "setup.py", line 90, in <module>
    'esptool.py',
  File "/usr/lib64/python2.7/distutils/core.py", line 111, in setup
    _setup_distribution = dist = klass(attrs)
  File "/usr/lib64/python2.7/site-packages/setuptools/dist.py", line 268, in __init__
    self.fetch_build_eggs(attrs['setup_requires'])
  File "/usr/lib64/python2.7/site-packages/setuptools/dist.py", line 313, in fetch_build_eggs
    replace_conflicting=True,
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 836, in resolve
    dist = best[req.key] = env.best_match(req, ws, installer)
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 1081, in best_match
    return self.obtain(req, installer)
  File "/usr/lib64/python2.7/site-packages/pkg_resources/__init__.py", line 1093, in obtain
    return installer(requirement)
  File "/usr/lib64/python2.7/site-packages/setuptools/dist.py", line 380, in fetch_build_egg
    return cmd.easy_install(req)
  File "/usr/lib64/python2.7/site-packages/setuptools/command/easy_install.py", line 638, in easy_install
    return self.install_item(spec, dist.location, tmpdir, deps)
  File "/usr/lib64/python2.7/site-packages/setuptools/command/easy_install.py", line 668, in install_item
    dists = self.install_eggs(spec, download, tmpdir)
  File "/usr/lib64/python2.7/site-packages/setuptools/command/easy_install.py", line 851, in install_eggs
    return self.build_and_install(setup_script, setup_base)
  File "/usr/lib64/python2.7/site-packages/setuptools/command/easy_install.py", line 1079, in build_and_install
    self.run_setup(setup_script, setup_base, args)
  File "/usr/lib64/python2.7/site-packages/setuptools/command/easy_install.py", line 1065, in run_setup
    run_setup(setup_script, args)
  File "/usr/lib64/python2.7/site-packages/setuptools/sandbox.py", line 240, in run_setup
    raise
  File "/usr/lib64/python2.7/contextlib.py", line 35, in __exit__
    self.gen.throw(type, value, traceback)
  File "/usr/lib64/python2.7/site-packages/setuptools/sandbox.py", line 193, in setup_context
    yield
  File "/usr/lib64/python2.7/contextlib.py", line 35, in __exit__
    self.gen.throw(type, value, traceback)
  File "/usr/lib64/python2.7/site-packages/setuptools/sandbox.py", line 152, in save_modules
    yield saved
  File "/usr/lib64/python2.7/site-packages/setuptools/sandbox.py", line 126, in __exit__
    self._saved = UnpickleableException.dump(type, exc)
  File "/usr/lib64/python2.7/site-packages/setuptools/sandbox.py", line 110, in dump
    return cls.dump(cls, cls(repr(exc)))
  File "/usr/lib64/python2.7/site-packages/setuptools/sandbox.py", line 110, in dump
    return cls.dump(cls, cls(repr(exc)))
  File "/usr/lib64/python2.7/site-packages/setuptools/sandbox.py", line 110, in dump
    return cls.dump(cls, cls(repr(exc)))
  File "/usr/lib64/python2.7/site-packages/setuptools/sandbox.py", line 110, in dump
[... 59 lines cut ...]
MemoryError
 * ERROR: dev-embedded/esptool-9999::quarks failed (compile phase):
 *   (no error message)
 * 
 * Call stack:
 *     ebuild.sh, line  133:  Called src_compile
 *   environment, line 4094:  Called distutils-r1_src_compile
 *   environment, line 1089:  Called _distutils-r1_run_foreach_impl 'distutils-r1_python_compile'
 *   environment, line  313:  Called python_foreach_impl 'distutils-r1_run_phase' 'distutils-r1_python_compile'
 *   environment, line 3638:  Called multibuild_foreach_variant '_python_multibuild_wrapper' 'distutils-r1_run_phase' 'distutils-r1_python_compile'
 *   environment, line 2743:  Called _multibuild_run '_python_multibuild_wrapper' 'distutils-r1_run_phase' 'distutils-r1_python_compile'
 *   environment, line 2741:  Called _python_multibuild_wrapper 'distutils-r1_run_phase' 'distutils-r1_python_compile'
 *   environment, line  774:  Called distutils-r1_run_phase 'distutils-r1_python_compile'
 *   environment, line 1080:  Called distutils-r1_python_compile
 *   environment, line  954:  Called esetup.py 'build'
[... cut ...]
 * If you need support, post the output of `emerge --info '=dev-embedded/esptool-9999::quarks'`,
 * the complete build log and the output of `emerge -pqv '=dev-embedded/esptool-9999::quarks'`.
 * The complete build log is located at '/var/log/portage/dev-embedded:esptool-9999:20160526-041403.log'.
 * The ebuild environment file is located at '/tmp/portage/dev-embedded/esptool-9999/temp/environment'.
 * Working directory: '/tmp/portage/dev-embedded/esptool-9999/work/esptool-9999'
 * S: '/tmp/portage/dev-embedded/esptool-9999/work/esptool-9999'

projectgus, 2016-05-26:
Weird. I'm not familiar with Gentoo, but can you explain the process (what commands were run) that got you to this point? I'm suspecting it's a problem with Gentoo's packaging process and setup_requires clauses in the setup.py, but I don't know enough either way to be sure.

Angus

projectgus, 2016-11-07:
@Quarky9 & @l29ah , are you still working on this? Any clues about ways to fix it in esptool.py's setup.py ?

projectgus, 2017-01-10:
I'm going to close this, esptool.py v2.0 has a lot of changes and you may have more luck re-running under Python 3. However feel free to reopen if you have any additional information that can help us solve this from the esptool side.

closed 2017-01-10 by projectgus