espressif/esptool #110: Specify difference of 32m, 32m-c1 and 32m-c2 values of --flash_size parameter

oyooyo, 2016-06-06:
I fail to understand the difference of the three possible choices for 32mbit flash modules that can be specified via the --flash_size option:
32m
32m-c1
32m-c2

It would be nice if there was a short mention of it in the documentation of the --flash_size option.

projectgus, 2016-07-14:
It's a good suggestion. Catch is, I don't fully understand these options either. They were added a long time ago, before I had anything to do with this project.

I believe they're used with OTA updates to save a region at the end of flash for user information, while the previous space is used for two program slots.

I believe 16m-c1 is what Espressif calls "512KB + 512KB", which means the 2MB of flash is split up as 512KB for one OTA slot, 512KB for another OTA slot, then 1MB for user data.

32m-c1 and 32m-c2 are for similar partitionings. I'm not entirely sure which one is which.

The whole thing is explained a bit in Espressif's PDF here:  http://bbs.espressif.com/download/file.php?id=1368

If someone figures this out, a pull request which adds the relevant lines to the README would be welcome!

closed 2017-05-03 by projectgus

simonliu009, 2019-01-12:
Below is my understanding:
Flash size 32Mbit: 512KB+512KB
Flash size 32Mbit-C1: 1024KB+1024KB

commit d2f35670 (2017-05-03): ESP8266: Add SPI flash sizes for 8MB, 16MB flash & document ESP8266 flash sizes

Closes #123
Closes #110