espressif/esptool #112: Fatal Error: arduino uno wifi

kruhly, 2016-07-15:
Using pip installed version 1.1, esptool.py generates:

A fatal error occurred: Failed to connect to ESP8266

for several of the read functions ( read_mac, flash_id , etc) when the Arduino1.7.10 IDE EspRecovery sketch is loaded onto Arduino Uno WiFi and the ESP8266 is put in bootloader mode.

The issue appears to be due to a single line of code introduced in commit:

24404b5f4f24221ee197cd4198f9731fbe11952e

as:

def __init__(self, port=0, baud=ESP_ROM_BAUD):
         self._port = serial.Serial(port)
+        self._slip_reader = slip_reader(port)
         # setting baud rate in a separate step is a workaround for
Removing self._slip_reader = slip_reader(port), resetting the arduino board, re-uploading the sketch and running esptool behaves as expected.

If setting self._slip_reader in the init is necessary for other functions, I believe is should call:

slip_reader(self._port)

as port is a string (untested).

projectgus, 2016-07-17:
Hi,

Thanks for the pointer about `slip_reader(port)`. You're right this should be `self._port`, and I've pushed a fix. However, as it happens the connect() method replaces self.slip_reader each time it makes a connection attempt, and it's initialised correctly there, so I don't know if this is the exact problem you're seeing in this case.

I don't know exactly what an "Arduino uno wifi" is, are you able to explain the hardware configuration a bit more please? It might be something special with the way the serial port is wired through.

Angus

kruhly, 2016-07-17:
Hi Angus,

It is an Arduino Uno with integrated ESP8266 wifi that my son received for his birthday. I don't know much about the architecture but there are tecnical spec's at http://www.arduino.org/products/boards/arduino-uno-wifi

Thanks,

Rob

projectgus, 2016-07-18:
Hi Rob,

I see, thanks for explaining.

When you say that you think slip_reader caused the problem, do you mean that older versions of esptool.py (like v1.0) work with the EspRecovery sketch?

Angus

kruhly, 2016-07-18:
Hi Angus,

I have not used v1.0 so no. 

The esp8266 was in a strange state due to an error when I tried to  change the static IP address in the Arduino Uno Wifi configuration. I was trying to recover it using the EspRecovery sketch in Arduino and esptool.py.

I was thinking it was the specific call to slip_reader in __init__ because I had seen the call in connect() -- possibly due the board not be ready. For some reason when I dropped the call to slip_reader in __init__ read_mem and other read functions started working.

However, I have now had some time to rerun  esptool.py with the esp8266 functioning normally. read_mem works as expected with the fixed call in __init__, without any call, and the bad call slip_reader(port). So it seems like it was a coincidence and  resetting the arduino board, finding the correct baud and/or  re-uploading the EspRecovery sketch  helped kick the esp8266 into normal operation again. 

Thanks for pushing the fix. 

Rob

projectgus, 2016-07-18:
Hi Rob,

Glad you got it sorted, these things can be finickity.

It sounds like I can close this issue now, but please let me know if that's not the case.

Angus

closed 2016-07-18 by projectgus