espressif/esptool #113: Upload not complete in Arduino IDE

Aizenko, 2016-07-19:
Hi,

On OSx i replace the default esp8266 esptool by the esptool.py. When uploading in arduino the upload stop at 110592 (48%); If I copy the command and run it in a shell it complete normally.

the command :
`opt/local/bin/python <path>/esptool.py -b 115200 -p /dev/cu.uart-F9FF4A1BBE192E4A write_flash -fm dio 0x00000 <path>/Blink.ino.bin`

projectgus, 2016-07-20:
Hi @Aizenko,

That's strange. I'm not sure what it could be, but maybe it has to do with buffering the output from stdout?

If you comment out the print statement `print ('\rWrote %d bytes at 0x%x in %.1f seconds ...` so it doesn't print as much status output, does it complete OK?

Angus

Aizenko, 2016-07-20:
I, You mean the lijne 
`sys.stdout.write(progress + '\b' * len(progress))`
I replace it by only a dot and got the same problem. But maybe the it's to long too.

Seem that the problem i got with the default esptools could be the same as it write dot, the something like "(37%) then a new line and dots (one third of the previous line, and the status bar shows "done uploading"... so it's may stop around 48% too.
I don't know how the arduino API detect the end of upload.

projectgus, 2016-08-03:
> I don't know how the arduino API detect the end of upload.

I expect it just waits for esptool.py (or whatever tool) to exit and checks the return code. I'm not sure I understand what you mean about the "default esptools", do you mean the esptool-ck tool that ships with Arduino for ESP8266?

projectgus, 2016-11-07:
@Aizenko , I'm going to close this because it's been a while and your last post suggests it may not be an esptool.py-specific problem. But feel free to reopen if you think esptool.py is misbehaving here.

closed 2016-11-07 by projectgus