espressif/esptool #115: Warning while creating bin file

romuye123, 2016-08-02:
I have used the below command to create bin file using .elf file.
`esptool.py elf2image my_app.elf`

Two .bin files get created named as follows:
`my_app.elf-0x00000.bin`
`my_app.elf-0x01010.bin`

But also get warning as:

`esptool.py v1.2-dev`
`WARNING: irom0 section offset is 0x00001010. ELF is probably linked for 'elf2image --version=2'`

What is it wrong i am doing? And how to upload these 2 files using esptool.py ?

projectgus, 2016-08-02:
If they were generate correctly, the binary files have the offsets that they should be flashed in the file names. So you can do write_flash 0x00000 my_app.elf-0x00000.bin 0x01010.bin my_app.elf-0x01010.bin

However, exactly what you flash where depends on what you used to build the ELF file in the first place. The project you used for building (was it Espressif's SDK or something else?) may have some instructions about flashing.

It's likely they want you to run "elf2image --version=2" instead, which will produce a single file instead of two files (which you then would usaully flash as `write_image 0x1000 my_app.bin` The warning about --version=2 is because esptool.py saw that the way the ELF file was structured internally (addresses of sections) implied that it may have been linked in a way to

romuye123, 2016-08-03:
Thanks for your reply @projectgus 
Tried executing the command 

`$ sudo esptool.py --port /dev/cu.SLAB_USBtoUART --baud 115200 write_flash 0x00000 my_app.cpp.elf-0x00000.bin 0x01010 my_app.cpp.elf-0x01010.bin`

and facing issues while uploading. Sometimes get time out and sometimes get assertion error. Following is the serial output:

`esptool.py v1.2-dev`
`Connecting...`
`Running Cesanta flasher stub...`

`A fatal error occurred: Timed out waiting for packet content`

OR

`AssertionError: Address must be sector-aligned`

projectgus, 2016-08-03:
Oh, right. Yes, I gave you bad advice - you almost certainly want to use elf2image --version=2 to create a single image to flash. You can only flash at addresses which are multiples of 0x1000.

The software environment that produced the ELF binary should have more instructions about what address they expect you to flash it at.

romuye123, 2016-08-03:
@projectgus : thanks for your reply.
I used the verison 2 command and the esptool created a single bin file. So it has created a bin file named `my_app-ota.bin`
So which address should i use to upload this bin file ?

`$ sudo esptool.py --port /dev/cu.SLAB_USBtoUART --baud 115200 write_flash 0x00000 my_app-ota.bin`

Is this correct ?

projectgus, 2016-08-03:
Hi,

I''ve suggested this twice already, but the software project that created the ELF file should have instructions about where you should flash it.

You'll need to flash some other bootloader binary at offset 0x0000, and then probably flash my_app-ota.bin at 0x1000. But I can't tell you this for sure, you need to go back to the project where you compiled your program from. Depending if it was Espressif's SDK or something else, it will have different requirements.

Angus

romuye123, 2016-08-05:
@projectgus 
I am using Arduino IDE for normal program upload,which works fine.
The arduino ide shows following 

`Uploading 365200 bytes from /Hexfiles/myapp.cpp.bin to flash at 0x00000000`

I believe arduino ide internally uses esptool-ck tool. 

I am confused on how to generate binary files using arduino ide and on which memory location the binary file should be uploaded using esptool.py?

projectgus, 2016-08-12:
I'm honestly not sure because I don't really use Arduino.

If you use the .bin file from Arduino (esptool-ck) then you can flash it at offset 0x0 in the same way (esptool.py --port port --baud baud write_flash 0 myapp.cpp.bin).

But "generate binary files" is a different problem. Do you want to go from an Arduuino .elf file to a .bin file without using esptool-ck?

projectgus, 2016-11-07:
Hi @romuye123 ,

Did you make any more progress with this issue?

Angus

projectgus, 2016-11-10:
Closing, but feel free to reopen if you think there's an esptool.py issue here.

closed 2016-11-10 by projectgus