espressif/esptool #117: Readme examples not accurate

PeterQFR, 2016-08-04:
Hi 
after trying to use esptool.py, the readme examples don't seem that accurate should be updated. I could only get a write flash command (with spi settings etc) to work only by specifying different arguments in different orders in the cli. 

for example the readme says to write to flash use:

```
esptool.py 0x01000 example.bin
```

but to actually achieve this in the majority of use cases you need

``
esptool.py --baud 115200 --port /dev/ttyUSB1 write_flash 0x00000 example.bin --flash_mode dio --flash_size 32m --flash_freq 40m
```

both the baud and port arguments need to predate `write_flash` and otherwise all SPI arguments need to suffix the .bin file. 

The documentation should be updated.

romuye123, 2016-08-04:
Agree with @PeterQFR 
even i am unable to flash correctly. many of the times i get following error:

`A fatal error occurred: Failed to connect to ESP8266`

projectgus, 2016-08-04:
@PeterQFR: fair point that a lot of these options were probably necessary but buried further down in the README. I've added some text near the top to draw attention to them. PRs that further clarify the README would be welcome.

@romuye123: Please go over the Troubleshooting steps listed in the README here: https://github.com/themadinventor/esptool#troubleshooting. If you still see errors then please open a new issue and explain your hardware setup and the steps you have tried.

commit f7678863 (2016-08-04): README: Clarify probably-necessary arguments for write_flash

Closes #117
closed 2016-08-04 by projectgus