espressif/esptool #118: Struggling to Flash with esptool

pieman64, 2016-08-07:
On a Windows 8.1 machine I can read the MAC and flash ID etc.
I can also backup the firmware but I am struggling to flash a bin file created in Arduino IDE.

Can the bin files from Arduino IDE be flashed with esptool?

projectgus, 2016-08-08:
Hi pieman64,

Should be possible, after the Arduino binary is generated it's no different from any other ESP8266 binary image.

Can you please post the command line you're using to flash, and some sample output? Have you been over the [Troubleshooting list in the README](https://github.com/themadinventor/esptool#troubleshooting)?

pieman64, 2016-08-08:
@projectgus thanks for your feedback and I have looked through the troubleshooting guide at length.

The device is an ESP07 and I'm using this comand:
**python esptool.py -p COM5 write_flash -fm dio -fs 8m 0x0000 Thermov3.0.0.bin**
```
esptool.py v1.2-dev
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0220
Writing 360448 @ 0x0... 0 (0 %)
A fatal error occurred: Timed out waiting for packet header
```
Just tried it with another ESP07 and it works fine as shown below:
```
python esptool.py -p COM5 write_flash -fm dio -fs 8m 0x0000 Thermov3.0.0.bin
esptool.py v1.2-dev
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0220
Writing 360448 @ 0x0... 360448 (100 %)
Wrote 360448 bytes at 0x0 in 31.7 seconds (90.9 kbit/s)...
Leaving...
```
So it looks like the first ESP07 is faulty. Tried to erase the flash as shown below but no joy. Is there anything else I can try?
```
python esptool.py -p COM5 erase_flash
esptool.py v1.2-dev
Connecting...
Erasing flash (this may take a while)...

A fatal error occurred: Timed out waiting for packet header
```

projectgus, 2016-08-09:
It does sound like some of your hardware is faulty.

You can try a lower baud rate, maybe 9600 (as suggest in the troubleshooting), to rule out any intermittent serial comms problems. I'd also double-check I was supplying adequate power to the modules, and double check all connections.

projectgus, 2016-11-07:
Closing this as it's been a while. If you think there's still an esptool.py issue here, please reopen.

closed 2016-11-07 by projectgus