espressif/esptool #122: 'A fatal error occurred: Timed out waiting for packet header' baudrate problem

Def152, 2016-09-04:
Hi

I saw other ppl that have this problem but didn't found anything except 'check your power supply'. I'm using ESP-12F powered from external 1A 3.3V power supply. 
When I'm flashing with the following command `python esptool.py --port /dev/ttyUSB0 write_flash --verify -fs 8m -fm dout 0x00000 image.elf-0x00000.bin 0x40000 image.elf-0x40000.bin` it stops at 99% while writting to 0x40000.
However if I try to flash with this command  `python esptool.py --port /dev/ttyUSB0 write_flash --verify 0x00000 image.elf-0x00000.bin 0x40000 image.elf-0x40000.bin` the bin is flashing with no problems. Also when i try to specify higher baudrate (eg 256000) I get `A fatal error occurred: Invalid head of packet ('\x92')`. 
I noticed it's only related to baudrate, `python esptool.py --port /dev/ttyUSB0 --baud 9600 write_flash --verify -fs 8m -fm dout 0x00000 image.elf-0x00000.bin 0x40000 image.elf-0x40000.bin` is working

I'm using CH340 donlge, I also tested it with CP2102 dongle. OS Linux Mint 18

projectgus, 2016-09-06:
Thanks for reporting this with so many details. The [Troubleshooting section for this problem](https://github.com/themadinventor/esptool#write_flash-operation-fails-part-way-through) gives two possibilities, one of which is power and the other one is to lower the baud rate. So it seems like you've gone over these already!

Do you get the exact same behaviour with the CP2102 as you do with the CH340? Is the ESP-12F on a breakout of some kind, or is it wired directly in? Any capacitors on/near the ESP-12F?

Def152, 2016-09-09:
The problem is on both, CP2102 and CH340. I'm using [this](http://s2.electrodragon.com/wp-content/uploads/2015/01/Wi12-breakout.jpg) breakout. Also there is 100nF cap on the back, close to ESP. Breakout board is connected via 10cm wires to UART adapter.

roger-zacharczyk, 2016-10-05:
@KaDw Have you been able to fix the problem?

projectgus, 2016-10-05:
@KaDw I missed this until roger commented before, but `-fs 8m` specifies a 1MB flash chip. If the image you're flashing is large, it may be that you're running out of flash. ESP-12Fs all have 4MB (32mbit) flash chips so you can use `-fs 32m` as a size argument.

projectgus, 2016-11-07:
Did you figure anything else out with this, @KaDw ? I'm tempted to think it's still a hardware issue of some kind (either power or USB/serial chip/drivers).

KaDw, 2016-11-09:
@roger-zacharczyk I used another power supply. I guess something happened to one that I used before.

@projectgus Yes, thats my bad, there should be -fs 32m. Today after a long break I tried again on another  power supply and it worked. As you mentioned it is probably hardware issue. Now even on 256000 baudrade it works like charm.

projectgus, 2016-11-10:
Good to hear, glad it's behaving now.

closed 2016-11-10 by projectgus

manchmal2, 2017-07-09:
Hey man. 
Im new to this and everytime i try to flash the nodemcu i get this: 
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
Flash params set to 0x0040
Compressed 461984 bytes to 298188...

A fatal error occurred: Timed out waiting for packet content
Im connecting it with a micro usb. Can you help me out?

roger-zacharczyk, 2017-07-09:
In my case it was a huge voltage drop between the left side and right side of the power lines in my breadboard. I switched to wires.

manchmal2, 2017-07-09:
Okay, i just connect it with the micro usb cable to my pc. Im not even able to use esplorer to communicate with it. How can i fix it?

ouaibsky, 2017-08-22:
Hi got exactly same issue with my ESP32 dev board

Someone do some progress ?

Regards
Chris

projectgus, 2017-08-22:
@ouaibsky I think everyone who reported this problem here found that it had something to do with power stability. Suggest going through the list of issues here: https://github.com/espressif/esptool/#troubleshooting

ouaibsky, 2017-08-23:
I check all things listed on troubleshooting page.

I'm using a macbook pro with usb - uart driver provided by https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

I plug USB from my macbook, voltage is qui stable: around 3.32 - 3.33V
When I push the "BOOT" Button on my dev board, I monitor GPIO0 and its becoming LOW (0.05V).

The strange things is I didn't get same output when trying to flash with Arduino or running esptool in a command line.

output from Arduino
```
esptool.py v2.1-beta1
Connecting....
Chip is ESP32D0WDQ6 (revision 0)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content
```

command line, it's worst, impossible to connect
```
e clalleme  ⎇  master  ~/Documents/Arduino/hardware/espressif/esp32/tools  2   ./esptool.py --port /dev/cu.SLAB_USBtoUART -b 9600 flash_id
sptool.py v2.1-beta1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
```

ouaibsky, 2017-08-23:
Correction, with command line I success to get flash_id
```
./esptool.py --port /dev/cu.SLAB_USBtoUART flash_id
esptool.py v2.1-beta1
Connecting........__
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 0)
Uploading stub...
Running stub...
Stub running...
Manufacturer: ff
Device: ffff
Detected flash size: Unknown
Hard resetting...
```

CaptClaude, 2017-08-25:
Here's another data point on flashing, ports and (perhaps) power.
I have a Mid-2014 MacBook Pro and was having trouble flashing MicroPython.
I did the normal thing, plugged my D1 Mini into a breadboard (with a blue power LED attached through a 220Ohm resistor), hooked up the USB cable _to the USB port on the left side of my computer,_ entered
`$ esptool.py --port /dev/tty.SLAB_USBtoUART --baud 460800 write_flash --flash_size=detect -fm dio 0 esp8266-20170823-v1.9.2.bin
esptool.py v2.1`
and got the dreaded `Connecting........_____.....(etc.)` then `...Timed out waiting for packet header`. 
Lowered eventually down to 9600 with no success. I took the ESP off the breadboard and tried again with the 9600.  Nothing.  Then I deleted the --baud and it worked.  Tried the USB port on the right side: worked.  Plugged the ESP back into the breadboard (still right USB port) and it worked. 
`$ esptool.py --port /dev/tty.SLAB_USBtoUART  write_flash --flash_size=detect -fm dio 0 esp8266-20170823-v1.9.2.bin
esptool.py v2.1
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0240
Compressed 601136 bytes to 392067...
Wrote 601136 bytes (392067 compressed) at 0x00000000 in 34.6 seconds (effective 138.9 kbit/s)...
Hash of data verified.
Leaving...
Hard resetting...`
With the ESP in the breadboard (with only one blue LED) I cannot flash with the left port, but I can with the right. I think a powered hub might help too.
1) Don't use the baud parameter
2) Different USB ports perform differently

roger-zacharczyk, 2017-08-25:
@CaptClaude 
Interesting. Out of curiosity: is there any difference between these two ports? Like USB 2.0 vs 3.0? Did you measure the voltage/current in both these ports?

CaptClaude, 2017-08-25:
I recall a long time ago reading that one port worked better for things like this but I can't find the reference anywhere.  A quick Google search shows that there is and has always been differences between the USB ports on Macs.  Why this would be true is anyone's guess.
I just tried reflashing with my little USB volt-ammeter plugged in and it won't flash from the right port with that in the circuit.  There was nothing unusual about the voltage or current (5.01V 89mA) while it was trying to get connected though.
I put down my ESPs for a while and picked them up again just recently so it's that old slow learning curve again.

fluentart, 2017-09-01:
I'm having this issue. It was able to upload to the board yesterday and this morning. When I tried to blink LED and add an interrupt function the board stopped responding to uploads with this same error. 

> __Build options changed, rebuilding all
> ___....._____....._____.....Archiving built core (caching) in: /var/folders/mm/x6q_vdnn7gz82s5n8dyj_kgm0000gn/T/arduino_cache_53890/core/core_espressif_esp32_esp32_FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none_21ef24fb8df1668ae86005e61e51fea4.a
> _Sketch uses 277607 bytes (21%) of program storage space. Maximum is 1310720 bytes.
> Global variables use 26736 bytes (9%) of dynamic memory, leaving 268176 bytes for local variables. Maximum is 294912 bytes.
> An error occurred while uploading the sketch
> _java.io.IOException: Stream closed
> 	at java.io.BufferedInputStream.getBufIfOpen(BufferedInputStream.java:170)
> 	at java.io.BufferedInputStream.read(BufferedInputStream.java:336)
> 	at sun.nio.cs.StreamDecoder.readBytes(StreamDecoder.java:284)
> 	at sun.nio.cs.StreamDecoder.implRead(StreamDecoder.java:326)
> 	at sun.nio.cs.StreamDecoder.read(StreamDecoder.java:178)
> 	at sun.nio.cs.StreamDecoder.read0(StreamDecoder.java:127)
> 	at sun.nio.cs.StreamDecoder.read(StreamDecoder.java:112)
> 	at java.io.InputStreamReader.read(InputStreamReader.java:168)
> 	at processing.app.debug.MessageSiphon.run(MessageSiphon.java:81)
> 	at java.lang.Thread.run(Thread.java:745)
> esptool.py v2.1-beta1
> Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____
> 
> A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
> A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

I'm on Mac Book Pro, with Arduino 1.8.3. And ESP-WROOM-32 by DOIT.AM (ESP32 DEVKIT V1)

ouaibsky, 2017-09-04:
Same result to me when using different port on my MacBook ( Think it's all USB3.0)

KaVo194, 2017-09-24:
Same issue with MacBook Pro 2017 via USB-C. Any update available?

projectgus, 2017-09-25:
It does seem like MacBooks are the common factor here. @ouaibsky @KaVo194 can you also please let me know:

* What ESP32 board/hardware do you have?
* If your board has a "BOOT" button, what happens if you hold this down while the flashing process starts?

ouaibsky, 2017-10-01:
I have issue with this one: https://www.aliexpress.com/item/ESP32-Development-Board-WiFi-Bluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32-ESP-32S-Board/32802136026.html

Just received this one and works like a charm: https://www.aliexpress.com/item/WEMOS-LOLIN32-V1-0-0-wifi-bluetooth-board-based-ESP-32-esp32-4MB-FLASH/32808551116.html

TrigerxD, 2018-08-24:
Hey all. I had the same problem "Timed out..." In my case some of straping pins had wrong logic state. Take multimeter and try that you have on gpio2 "0" gpio0 "0" and gpio12 "0". That last decide that at '0' memory is at 3.3V (what we want) and at '1' at 1.8V what causes many problems.
Here is great explanation how it should be 
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

racielrod, 2018-09-02:
If you are using a MacBook - as some of you have reported - make sure your charger is plugged in. After connecting the charger to the macbook I was able to flash w/o issues.

Apaezmx, 2019-01-05:
While waiting for my external 3.3v power supply to arrive, I played around with esptool timeouts. Seems if you extend the timeout constants within the py file, and set a low baud (4800) it works with Arduino UNO 3.3v supply. The only downside is that it takes longer to do stuff.

projectgus, 2019-01-06:
@Apaezmx, that's a very interesting result! I think the bad news will be if you try to boot the ESP8266 and actually do anything (like connect to WiFi), it will immediately brown out and result or end up in a bad state.

Apaezmx, 2019-01-06:
Hmm... why?
It has worked so far. I've been using my modified esptool.py to program my ESP8266 for a few days now. No issues other than flashing takes 1-2 minutes.

projectgus, 2019-01-06:
@Apaezmx Flashing uses a small amount of power, but enabling the WiFi radio and transmitting uses a much larger amount of power.

If it's working for you then that's great.

Apaezmx, 2019-01-07:
@projectgus I see. Then I guess my issue was not with the power supply :).

anandkilli, 2019-05-17:
Got the same error as follows and my issue was power. Connect 5v and Gnd from bread board power supply and I am good
"A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header"

15754601041, 2020-07-30:
> 嗨，老兄。
> 我是新来的，每次尝试刷新nodemcu时我都会得到：正在
> 连接 ..._ 正在
> 检测芯片类型... ESP8266
> 芯片为ESP8266
> 正在上传存根...正在
> 运行存根...
> 正在存根运行..正在
> 配置闪存大小...
> 警告：无法自动检测闪存大小（FlashID = 0x0，SizeID = 0x0），默认
> 设置为4MB 闪存参数设置为0x0040
> 压缩461984字节为298188 ...
> 
> 发生致命错误：超时，等待数据包内容
> Im连接到微型USB。你能帮我吗？

查看IO12的电平，3.3  or  1.8 ？？？？！！！！