espressif/esptool #123: support for W25Q128 flash chip

stephaneAG, 2016-09-08:
Hi

I had few ESP-01 handy as well as a Winbond W25Q128 flash chip, so I decided to give the flash replacement procedure a try before running on it the latest Espruino firmware ( .. )

The last step being actually flashing the chip, I'm now wondering how to mod esptool.py to support it ( .. )

If any hint on those, feel free to give me an update ;)

https://github.com/stephaneAG/EspruinoDocs/blob/master/tutorials/ESP8266_FlashUpgrade.md

keep up the good work ;p
+

projectgus, 2016-09-09:
Hi stephane,

Interesting hack!

I think from the esptool.py side it should more or less "just work" as a 32mbit chip. The flash-size parameter is for the software & ROM routines on the ESP to know how large the flash chip is. 32mbit (`-fs 32m`) is the largest supported by the Espressif SDK (and possibly also the the ROM bootloader will check this value). There are no other constant values supported by the SDK to indicate larger flash sizes.

So, to begin with, I'd suggest just flashing with `-fs 32m` and make sure it works that much.

Regarding actually using the extra 96Mbits of flash, I believe all of the ROM SPIErase/SPIWrite/etc routines check a data structure in RAM called "flashchip" (initialised by the ROM) to determine the flash size and whether an operation is out of bounds. You could try overriding this value to trick the ROM routines into letting you attempt operations on higher offsets.

However, I also suspect (I don't know for sure) that the SPI hardware may not support addressing beyond the number of address bits required for 32Mbit. So requesting past this address may just wrap the actual SPI operation, so it returns back to the beginning.

If this turns out to be the case, you could always write your own SPI access routines that talk to the hardware directly instead of via the ROM functions...

If you have a logic analyzer, tracing the SPI flash commands while messing with the bootloader stub is probably a place to start. That way you can see exactly what is going out to the chip each time.

Good luck! :)

Angus

projectgus, 2016-11-07:
Hi @stephaneAG ,

How did you go with this? Is there anything you think we can/should add in esptool.py to support oversize flash of this kind? Otherwise I might close this issue.

Angus

pfalcon, 2016-11-08:
There're now off-shelf modules with more than 4MB of flash. I'm in particular concerned with how auto-detection works on them - it's rather unpleasant (for a user) when his 16MB device treated as 0.5MB. That's why implemented auto-detection the I did originally, so any unknown flash ID leads to error with all info which a user can report and get resolution. The way you requested it to be changed, auto-detection is not reliable and can give a user headache, as well as anyone who supports software on wide variety of modules.

I'm going to look into this, but it probably will take as much time as between auto-detection RFC and actual implementation.

projectgus, 2016-11-08:
Do you have some URLs or flash ids for the modules which are shipping with 16MB?

pfalcon, 2016-11-08:
Wemos Pro or how they called it. It now lies in my board dump along with dozens of other boards and in few months I'll get to power it up and read the ID.

mcauser, 2016-11-09:
WeMos D1 Mini Pro:
https://www.wemos.cc/product/d1-mini-pro.html
Features a WinBond W25Q128FVSG.

```
$ esptool.py --port /dev/tty.SLAB_USBtoUART -b 115200 flash_id
esptool.py v1.2.1
Connecting...
Manufacturer: ef
Device: 4018
```

marcelstoer, 2016-12-19:
> 32mbit is the largest supported by the Espressif SDK

I was told that according to the ESP8266EX data sheet it supports max 128mbit. The advent of those 16MB WeMos modules kind-of proves that.

The NodeMCU firmware project still hosts its own copy of `esptool.py` (don't ask, I don't like it). It has had partial support for `64m` and `128m` since June: 
https://github.com/nodemcu/nodemcu-firmware/pull/1367/commits/7ecd625db6c21b2ce4122adcb0993d677ccd9877#diff-e6e8c9cf42985f99554a22a8a34b90e4 I say partial because I don't see how that quick-fix would play nice with auto-size detection.

sandmarq, 2017-02-02:
I have the same WeMos d1 mini pro and I was wondering if anyone had any plan to look at this issue.

> ./esptool.py --port /dev/ttyUSB0 -b 115200 flash_id
esptool.py v2.0-beta1
Connecting....
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4018
Hard resetting...

I was able to flash it by forcing it to use only 4M of the 16M.

marcelstoer, 2017-02-11:
Any chance this will be included in 2.0?

projectgus, 2017-02-13:
> Any chance this will be included in 2.0?

Sure. If the ESP8266 ROM bootloader is happy with these new flash size header values (0x80 and 0x90) then we can add these as well. If that's the case, could you please send a pull request which adds these same values as used in nodemcu's fork?

devsaurus, 2017-02-17:
Current master correctly detects the 16MB flash but aborts with an error message that this size isn't supported by the chip. This seems to be too restrictive, but are these new flash size values (0x80 and 0x90) supported by the SDK? IMO there's not much use in introducing new values if the SDK can't determine the location of the init data from them.

AFAIK the existing solutions for Android and NodeMCU apply some kind of tweak to use the module in a "known good" configuration like 4MB or 1MB. Guess a 100% clean approach needs support from the SDK for 8/16MB, anything else is a workaround.

My proposal for "8/16MB support by esptool.py on ESP8266" is to turn the error-and-abort condition when >4MB is detected into a warning and set the flash size to e.g. 1MB automatically. Any tweaks in the firmware are still required for sure - just the flash autodetection shouldn't block downloading anymore.

Any objections / additions? I'd come up with a respective PR.

marcelstoer, 2017-02-17:
If you replace "Android" with "Arduino" (my guess) then this sounds reasonable 😆.

 > turn the error-and-abort condition when >4MB is detected into a warning and set the flash size to e.g. 1MB automatically

How likely is it that this would upset _some_ firmwares? Do you know if there are cases where a firmware would fail if esptool.py used 1MB instead of 4MB (or vice versa)?

devsaurus, 2017-02-17:
> If you replace "Android" with "Arduino"

😊 

> How likely is it that this would upset some firmwares?

Don't know, this depends on the specific firmware. If a FW relies exclusively on the flash size header it should adapt correctly. If it tries to be smart and inspects the true physical size of the flash it might get into trouble.
This all boils down to the question how to deal with a hardware configuration that's not specified by Espressif and thus isn't backed by consistent SDK support.

Current esptool.py takes the safe route and rejects >4MB: No service for an unspec'd HW configuration. The user can always override this with the `-fs` option, though. If he knows what he does.

marcelstoer, 2017-02-17:
> how to deal with a hardware configuration that's not specified by Espressif and thus isn't backed by consistent SDK support.

It _is_ specified, chapter 3.1.3 specifies 16 MB max, even though it says "theoretically": http://espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf#page=12

devsaurus, 2017-02-17:
Right, the hardware is able to deal with 16MB.

Different for the software: Chapter 4 in http://espressif.com/sites/default/files/documentation/2a-esp8266-sdk_getting_started_guide_en.pdf doesn't specify flash maps for >4MB.

igrr, 2017-02-21:
We'll update the ESP8266 non-OS SDK to support 16MB flash size in the next release (roughly in a month).

dwery, 2017-02-24:
FYI, https://github.com/igrr/esptool-ck was able to flash my D1 pro correctly

marcelstoer, 2017-03-28:
@igrr out of curiosity...how are things developing wrt that next release?

capedra, 2017-04-09:
@igrr Hello Ivan. Any news about supporting the 16 MB flash size from WeMos D1 Mini Pro?

projectgus, 2017-04-09:
Hi @marcelstoer & @capedra ,

The ESP8266 non-OS SDK release with support for 8MB & 16MB flash sizes is being tested now, ahead of a release. esptool.py support for these sizes should be available in the master branch shortly.

Angus

projectgus, 2017-04-24:
Hi everyone,

Thanks for bearing with us, I know this process has been a bit slow.

Changes for 8MB & 16MB flash sizes are in d2f35670, ready to merge. @marcelstoer (and anyone else), can you please confirm that all works for you? There's a new note in the README about the 4MB-c2 configuration, which I think was added for NodeMCU some years ago (it's not supported in Espressif's SDKs) - is that correct?

Cheers,

Angus

marcelstoer, 2017-04-24:
> Changes for 8MB & 16MB flash sizes are in d2f3567, ready to merge.

Looking forward to it. Will you make a 1.3.1 release that contains this fix or will we have to wait for 2.0-final? Reason for asking: my policy for the [PyFlasher](https://github.com/marcelstoer/nodemcu-pyflasher) is that I only incorporate final esptool.py releases.

> 4MB-c2 configuration, which I think was added for NodeMCU some years ago

I wouldn't know, sorry.

> The 2.1.0 ESP8266 non-OS SDK to support these flash sizes will be released shortly.

Will we see that on https://github.com/espressif/ESP8266_NONOS_SDK?

pieman64, 2017-04-24:
@projectgus the 8MB and 16MB update was committed 18 days ago but are you saying it's not been merged yet and therefore not available? Is there any way we can start using this now and if so how?
We want to use it from the Arduino IDE.

capedra, 2017-04-24:
@igrr @projectgus 
everyone, please, help us on [here](https://github.com/platformio/platform-espressif8266/issues/8)

igrr, 2017-04-24:
@pieman64 @capedra To get 8MB and 16MB working properly in Arduino, esptool.py update is not necessary, as we are using [a different tool](https://github.com/igrr/esptool-ck) to generate images and do flashing. But we do need an update to the non-OS SDK, which will soon be released.

Once the updated SDK is available, I will implement 8/16MB support in the Arduino core. Then the hacked mentioned in that platformio discussion will not be needed.

capedra, 2017-04-24:
@igrr Hello Ivan, thanks a lot for your support. I can see that several people do not only want but they actually need this 16MB support as soon as possible. Thousands of people who buy WeMos D1 mini Pro.

projectgus, 2017-04-24:
> Looking forward to it. Will you make a 1.3.1 release that contains this fix or will we have to wait for
> 2.0-final? Reason for asking: my policy for the PyFlasher is that I only incorporate final esptool.py releases.

A reasonable policy! I'm hoping to only do bugfix releases to v1.x from now on. However, what I can do is fast-track v2.0 final. It's been a stable beta for a while now and the amount of remaining tidy-up work is small (mostly testing the new espefuse/espsecure command line tools work fully with Python 3.)

>>    4MB-c2 configuration, which I think was added for NodeMCU some years ago
> I wouldn't know, sorry.

OK, no problems. Looking at the history of this again, looks like it was added in #42, and the PR contributor mentioned Espressif SDK v1.0.2. However support for this size appears to have never been added to the SDK build tools back then, which is very odd. So I'll mark it as fully unsupported.

>>    The 2.1.0 ESP8266 non-OS SDK to support these flash sizes will be released shortly.
> Will we see that on https://github.com/espressif/ESP8266_NONOS_SDK?

Yes, the SDK release has been delayed due to final testing of some changes (unrelated to this change), but as soon as that testing is complete it will be released.

closed 2017-05-03 by projectgus

capedra, 2017-05-06:
@igrr @projectgus Hello software engineers! As the updated SDK became available earlier today, can the colleagues please implement 8/16MB support in the Arduino core? Thank you very much!!

marcelstoer, 2017-05-06:
Two down (SDK release, esptool commit), one more to go. Any idea when v2.0 is gonna be released?

capedra, 2017-05-17:
@igrr Any news about your implementation?

igrr, 2017-05-17:
@capedra you can go to https://github.com/esp8266/Arduino/issues/3215 and click "Subscribe" to get updates :)

commit d2f35670 (2017-05-03): ESP8266: Add SPI flash sizes for 8MB, 16MB flash & document ESP8266 flash sizes

Closes #123
Closes #110