espressif/esptool #125: Access rights to /dev/ttyUSB*

frenzymadness, 2016-09-14:
Hello.

I and my colleagues are working on stuff related to MicroPython in Fedora. We created a package with esptool [0] page about MicroPython for Fedora developers portal etc.

Because we are trying to make all of this stuff as simple as possible for developers to use I have one question: Is there known a simple way how a user can gain privileges to access connected device (/dev/tty*)?

Known solutions are:
* Use `sudo` (unsafe)
* Add regular user account to *dialout* system group

We are thinking about creating new special package which will contain only udev rules file and make dependency between the esptool package and new *esp-udev-rules* package. This change gives access rights to users automatically after installation of esptool from Fedora repositories.

What do you think about it?

[0] https://apps.fedoraproject.org/packages/esptool

hroncok, 2016-09-14:
See https://github.com/hroncok/3dprinter-udev-rules/blob/master/66-3dprinter.rules as an example how this could be achieved. It requires gathering PIDs and VIDs of the devices.

encukou, 2016-09-14:
I guess the question is: Could such rules be part of the esptool repo, where all Linux distros/users can take advantage of them?

projectgus, 2016-09-15:
Hi folks,

Thanks for getting in touch. I agree the permissions issue on Linux is a tricky one. At the very least I guess it should be mentioned in the README for esptool.py as well.

Is there a downside of telling users about the dialout group? I guess if it's incorporated in the package then it's seamless rather than requiring a manual command & a re-login.

My concern with collecting udev rules is that all of the esp8266 hardware I know about uses generic USB/serial adapter chip USB PIDs/VIDs - the common chips are FT232R, CP2102, CH340G but more or less every USB/TTL adapter is being used sometimes. So a comprehensive set of udev rules to cover all of these would amount to covering all current generation USB/serial adapters. Which seems fairly close to what the "dialout" group is designed to handle - "on this system, can this user access serial ports?"

That said, if you were to submit a PR with a set of udev rules like that then I'd be happy to host them in this repo and mention them as an option in the README. I'm guessing they'd work with any systemd-enabled distro (the uaccess tag is a udev-systemd thing, yes?)

BTW, it's awesome you're packaging esptool into Fedora. Much appreciated! :)

frenzymadness, 2016-09-15:
Hello.

The downside of a dialout group is that to add yourself to it, you need to run a command and then log out & in. We're trying to streamline the experience for newcomers.
 
Your argument about dialout is fair. We'll investigate adding users there automatically. If that doesn't work out, we'll send a pull request. Thanks for your willingness to accept it!
The linked rules are systemd-specific, yes.
 
Thanks for writing esptool!

projectgus, 2016-09-15:
OK @frenzymadness,, sounds good

If it's OK with you, I'm going to leave this issue open as a reminder that the esptool README should mention permissions on Linux.

frenzymadness, 2016-09-16:
ok, thank you

commit 236db9a1 (2017-12-29): README: Add note about "dialout" group on Linux

Closes #125
closed 2017-12-29 by projectgus