espressif/esptool #129: Suggestion: Rename "Unc size" and "comp size"

nkolban, 2016-10-07:
Please file this under incredibly small stuff ....  when flashing to the ESP32 I have enabled compression and it is working great.  However the message issued by the tool is "not to my liking".  The message is written at:

https://github.com/themadinventor/esptool/blob/197ba605fe0c05e16bf4c5ec07b726adc8d86abc/esptool.py#L423

and for me says:

`Unc size 217408 comp size 59054 comp blocks 58`

What I would prefer it to say is:

`Uncompressed size 217408,  Compressed size 59054 comp blocks 58`

I realize how trivial this is ... so please feel free to close with "no" at any time ... :-)

projectgus, 2016-10-07:
It's a good request. The esptool.py for ESP32 is still a work-in-progress, the full documentation for its status is here:
https://github.com/themadinventor/esptool/pull/121

One of the "TODOs" in the back of my mind was to finish cleaning up the output (which includes some debug strings like this one.) I'll put it on the official todo list, as well.

projectgus, 2016-11-07:
This is now fixed in the esp32 branch (ref #121):

```
esptool.py v2.0-dev
Connecting...
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
Compressed 282836 bytes to 123726...
Wrote 282836 bytes (123726 compressed) at 0x00010000 in 10.9 seconds (effective 207.8 kbit/s)...
Hash of data verified.
Compressed 3140 bytes to 149...
Wrote 3140 bytes (149 compressed) at 0x00008000 in 0.0 seconds (effective 1418.6 kbit/s)...
Hash of data verified.
```

closed 2016-11-07 by projectgus