espressif/esptool #130: DTR

serus22, 2016-10-07:
hello
im trying to run esptool on raspberry using rpi uart, but it never connects it
in section Automatic bootloader is this truth ?: "Note that some serial terminal programs (not esptool.py) will assert both RTS and DTR when opening the serial port..."
and then in code u using as:
https://github.com/themadinventor/esptool/blob/master/esptool.py#L134
so, if i want to use rpi to flash esp, shouuld i change this to another rpi GPIO?, am i wrong or right?

thanks

projectgus, 2016-10-09:
How is your ESP-12F wired to the Raspberry Pi serial port (ie which pins are wired up)?

serus22, 2016-10-10:
same as https://ae01.alicdn.com/kf/HTB114xvJFXXXXcDXFXXq6xXFXXXj/120336020/HTB114xvJFXXXXcDXFXXq6xXFXXXj.jpg but i tryied to connect gpio0 to gnd/uart-rts/+3.3  and its still not working, also external esp power source with connected gnds.

projectgus, 2016-10-10:
In addition to the diagram you posted, you also need a way to reset the chip into the bootloader (by toggling RST or CH_PD low then high.)

Suggest trying the following steps:

* Tie GPIO0 to ground permanently
* Open a serial terminal at 74880 bps (Not all serial programs can do 74880. You can run `miniterm.py /dev/ttySx 74880`.) and leave that running.
* Pull CH_PD low and then release to let the pull-up pull it back to VCC.

You should see some output printed on the miniterm console. If you don't, it indicates you still have a wiring problem.

If the above works, quit miniterm.py and run esptool.py (don't touch the ESP module wiring in between these two steps), and see if that works.

Angus

serus22, 2016-10-18:
Srorry for late reply, here its result of miniterm.py followed by 2 attempts of flash by esptool
http://prnt.sc/cvy7gb

> ets Jan  8 2013,rst cause:2, boot mode:(5,5) waiting for host

was printed to console when i push reset button

bvernoux, 2016-10-18:
For full details on this issue I have done a full analysis of the problem on Win7 see
http://esp32.com/viewtopic.php?f=2&p=1511#p1511

projectgus, 2016-10-19:
(@serus22, I think @bvernoux's post is intended for the other issue regarding ESP32+Windows, not this issue. You can probably skip it.)

```
ets Jan 8 2013,rst cause:2, boot mode:(5,5) waiting for host
```

`boot mode:(5,` and `waiting for host` indicate the ESP8266 has gone into SDIO boot mode. You need to pull GPIO15 low to avoid this mode. There's a full description of the pin configuration needed here:
https://github.com/themadinventor/esptool#required-pins

bvernoux, 2016-10-19:
I have moved my comment to https://github.com/themadinventor/esptool/issues/136 as it was not linked to this issue

serus22, 2016-10-19:
today i tryed this converter
http://www.ebay.com/itm/121697917486?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

wired as here 
http://prnt.sc/cwfnbx

so the result is: flashing is finished successfully (by esp flasher), but if i reset module, open esplorer and push connect button, module start sending infinity unknown data stream, and led its blinking, if i reset r power of module and start again, esp continue in sending unknown data and blinking, i can stop it by pressing Flash button and pressing rst or enable pulldown button. If im doing whatever, its stil just in these 2 states. i think flashing isnt working bcs if i connect via putty with 74880 baudrate, there is always same message "ets Jan....." 
im rly confused.
i tryed a lot of combinations but nothing works

pfalcon, 2016-10-19:
There're definitely weird modules around. For example, one of my modules started (??) to exhibit behavior that after few resets (e.g. because of flashing), it's stuck after printing "ets Jan....." line. Pressing reset doesn't help. The module needs to disconnected from USB and reconnected again. With increasing chance, single reconnect doesn't help. This module uses CH341 USB converter.

closed 2016-10-26 by serus22

projectgus, 2016-10-26:
@serus22, are you closing because you got this working or did you give up?

serus22, 2016-10-26:
sorry for no comment, i am closing it because i do not think it is general problem but its not resolved

projectgus, 2016-10-26:
No worries. It does sound like a hardware-specific problem. Good luck figuring it out!