espressif/esptool #132: error: too few arguments.....

screig, 2016-10-09:
```
C:\WINDOWS\system32>esptool.py -H
usage: esptool [-h] [--port PORT] [--baud BAUD]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash,verify_flash,erase_flash,version}
               ...
esptool: error: too few arguments

C:\WINDOWS\system32>esptool.py -h
usage: esptool [-h] [--port PORT] [--baud BAUD]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash,verify_flash,erase_flash,version}
               ...
esptool: error: too few arguments

C:\WINDOWS\system32>esptool.py -help
usage: esptool [-h] [--port PORT] [--baud BAUD]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash,verify_flash,erase_flash,version}
               ...
esptool: error: too few arguments

C:\WINDOWS\system32>esptool.py -HELP
usage: esptool [-h] [--port PORT] [--baud BAUD]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash,verify_flash,erase_flash,version}
               ...
esptool: error: too few arguments

C:\WINDOWS\system32>esptool.py -version
usage: esptool [-h] [--port PORT] [--baud BAUD]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash,verify_flash,erase_flash,version}
               ...
esptool: error: too few arguments

C:\WINDOWS\system32>esptool.py --version
usage: esptool [-h] [--port PORT] [--baud BAUD]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash,verify_flash,erase_flash,version}
               ...
esptool: error: too few arguments

C:\WINDOWS\system32>

```

projectgus, 2016-10-09:
That's odd! Can you post the output of `python -V`, please?

projectgus, 2016-10-09:
(FWIW, the arguments that should be recognised from the ones you posted are `esptool.py -h` and `esptool.py --help`.)

screig, 2016-10-10:
```
C:\WINDOWS\system32>python -v
# installing zipimport hook
import zipimport # builtin
# installed zipimport hook
# C:\tools\python2\lib\site.pyc matches C:\tools\python2\lib\site.py
import site # precompiled from C:\tools\python2\lib\site.pyc
# C:\tools\python2\lib\os.pyc matches C:\tools\python2\lib\os.py
import os # precompiled from C:\tools\python2\lib\os.pyc
import errno # builtin
import nt # builtin
# C:\tools\python2\lib\ntpath.pyc matches C:\tools\python2\lib\ntpath.py
import ntpath # precompiled from C:\tools\python2\lib\ntpath.pyc
# C:\tools\python2\lib\stat.pyc matches C:\tools\python2\lib\stat.py
import stat # precompiled from C:\tools\python2\lib\stat.pyc
# C:\tools\python2\lib\genericpath.pyc matches C:\tools\python2\lib\genericpath.py
import genericpath # precompiled from C:\tools\python2\lib\genericpath.pyc
# C:\tools\python2\lib\warnings.pyc matches C:\tools\python2\lib\warnings.py
import warnings # precompiled from C:\tools\python2\lib\warnings.pyc
# C:\tools\python2\lib\linecache.pyc matches C:\tools\python2\lib\linecache.py
import linecache # precompiled from C:\tools\python2\lib\linecache.pyc
# C:\tools\python2\lib\types.pyc matches C:\tools\python2\lib\types.py
import types # precompiled from C:\tools\python2\lib\types.pyc
# C:\tools\python2\lib\UserDict.pyc matches C:\tools\python2\lib\UserDict.py
import UserDict # precompiled from C:\tools\python2\lib\UserDict.pyc
# C:\tools\python2\lib\_abcoll.pyc matches C:\tools\python2\lib\_abcoll.py
import _abcoll # precompiled from C:\tools\python2\lib\_abcoll.pyc
# C:\tools\python2\lib\abc.pyc matches C:\tools\python2\lib\abc.py
import abc # precompiled from C:\tools\python2\lib\abc.pyc
# C:\tools\python2\lib\_weakrefset.pyc matches C:\tools\python2\lib\_weakrefset.py
import _weakrefset # precompiled from C:\tools\python2\lib\_weakrefset.pyc
import _weakref # builtin
# C:\tools\python2\lib\copy_reg.pyc matches C:\tools\python2\lib\copy_reg.py
import copy_reg # precompiled from C:\tools\python2\lib\copy_reg.pyc
# C:\tools\python2\lib\traceback.pyc matches C:\tools\python2\lib\traceback.py
import traceback # precompiled from C:\tools\python2\lib\traceback.pyc
# C:\tools\python2\lib\sysconfig.pyc matches C:\tools\python2\lib\sysconfig.py
import sysconfig # precompiled from C:\tools\python2\lib\sysconfig.pyc
# C:\tools\python2\lib\re.pyc matches C:\tools\python2\lib\re.py
import re # precompiled from C:\tools\python2\lib\re.pyc
# C:\tools\python2\lib\sre_compile.pyc matches C:\tools\python2\lib\sre_compile.py
import sre_compile # precompiled from C:\tools\python2\lib\sre_compile.pyc
import _sre # builtin
# C:\tools\python2\lib\sre_parse.pyc matches C:\tools\python2\lib\sre_parse.py
import sre_parse # precompiled from C:\tools\python2\lib\sre_parse.pyc
# C:\tools\python2\lib\sre_constants.pyc matches C:\tools\python2\lib\sre_constants.py
import sre_constants # precompiled from C:\tools\python2\lib\sre_constants.pyc
import _locale # builtin
# C:\tools\python2\lib\locale.pyc matches C:\tools\python2\lib\locale.py
import locale # precompiled from C:\tools\python2\lib\locale.pyc
import encodings # directory C:\tools\python2\lib\encodings
# C:\tools\python2\lib\encodings\__init__.pyc matches C:\tools\python2\lib\encodings\__init__.py
import encodings # precompiled from C:\tools\python2\lib\encodings\__init__.pyc
# C:\tools\python2\lib\codecs.pyc matches C:\tools\python2\lib\codecs.py
import codecs # precompiled from C:\tools\python2\lib\codecs.pyc
import _codecs # builtin
# C:\tools\python2\lib\encodings\aliases.pyc matches C:\tools\python2\lib\encodings\aliases.py
import encodings.aliases # precompiled from C:\tools\python2\lib\encodings\aliases.pyc
import operator # builtin
# C:\tools\python2\lib\functools.pyc matches C:\tools\python2\lib\functools.py
import functools # precompiled from C:\tools\python2\lib\functools.pyc
import _functools # builtin
# C:\tools\python2\lib\encodings\cp1252.pyc matches C:\tools\python2\lib\encodings\cp1252.py
import encodings.cp1252 # precompiled from C:\tools\python2\lib\encodings\cp1252.pyc
# zipimport: found 132 names in C:\tools\python2\lib\site-packages\setuptools-26.1.1-py2.7.egg
Python 2.7.10 (default, May 23 2015, 09:44:00) [MSC v.1500 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>>
```

projectgus, 2016-11-24:
Hi,

Sorry I never got back to you, to be honest this behaviour has me a bit stumped - the argument parsing functionality built into Python, and your Python installation looks correct to me.

One more thing to try if you're still trying to figure this out, use a text editor to add two lines in esptool.py after the first line, like this:

```
#!/usr/bin/env python                                                                                                                                                        
import sys                                                                                                                                                                   
print (sys.argv, sys.platform, sys.version_info)                                                                                                                             
# NB: Before sending a PR to change the above line to '#!/usr/bin/env python2', please read https://github.com/espressif/esptool/issues/21                                   
#                                                                                                                                                                            
# ESP8266 ROM Bootloader Utility     
```

And run `esptool.py -h` again, you should get a line with some more details about the current running Python environment (and what it thinks the command line arguments were.)

visasimbu, 2016-12-19:
Include python word in front of your command. So your command should something like below.

C:\WINDOWS\system32>python esptool.py --version

projectgus, 2017-01-10:
@screig , did you have any luck following @visasimbu's suggestion?

projectgus, 2017-12-29:
Closing due to lack of updates. Feel free to reopen if this is still an issue.

closed 2017-12-29 by projectgus