espressif/esptool #133: ESP_FLASH_BLOCK

YorkJong, 2016-10-11:
I thinks that ESP_FLASH_BLOCK should be 0x1000 (not 0x400)

projectgus, 2016-10-19:
ESP_FLASH_BLOCK doesn't get used for very much now that the flasher stub is used since esptool v1.1. Do you have some more rationale for why it should be higher?

YorkJong, 2016-10-19:
I don't know why the value of ESP_FLASH_BLOCK is 0x400. As I known, almost all serial (NOR) flashes support 64k  block size (the erasure unit), some serial flashes also support 32k block size, and a few serial flashes additionally support 4k erasure unit, so called *sector*. None of them support 0x400 (1k) erasure unit.

projectgus, 2016-10-19:
Ah, I see. The "block" referred to by ESP_FLASH_BLOCK is part of the serial protocol of the ROM bootloader (data gets sent as "blocks" in the protocol). It's not related to the block erase command of the SPI flash.

closed 2016-10-19 by YorkJong