espressif/esptool #134: Divide by zero errors

negativekelvin, 2016-10-13:
https://github.com/themadinventor/esptool/blob/197ba605fe0c05e16bf4c5ec07b726adc8d86abc/esptool.py#L1314

https://github.com/themadinventor/esptool/blob/197ba605fe0c05e16bf4c5ec07b726adc8d86abc/esptool.py#L1436

t can be zero, causing a divide by zero error.  May vary based on the precision of the float value returned by time()

projectgus, 2016-10-13:
Good catch, thanks. I'm guessing you're seeing this on some version of Windows?

I think time.clock() is only useful on Windows, it's process CPU time on macOS & Linux. In Python 3.3 they added time.perf_counter() to work around this.

In any case, we can fix the divide by zero error.

negativekelvin, 2016-10-13:
Yes this was on windows 7 with esp32, compressed uploads, and 921600 baud.  All those factors can lead to t <100ms which results in zero value.

projectgus, 2016-11-04:
This is fixed in the v2.0 ESP32 development branch. I haven't backported the fix to master as I think it's probably only a problem with compressed upload turned on, but I'll close this once it's merged there.

projectgus, 2017-01-10:
esptool.py v2.0 is now merged to master, so this fix is also in master.

closed 2017-01-10 by projectgus